raspberrypi/Raspberry-Pi-OS-64bit #1: there is some package problem when I want to install gnome environment with tasksel

xlazom00, 2020-05-28:
There is some package problem when I want to install gnome environment with tasksel

solskogen, 2020-05-28:
It seems like libraspberrypi-dev conflicts with a couple of packages.

ghollingworth, 2020-05-28:
It's probably related to us moving the /opt/vc contents into their rightful and correct places (/usr/lib for example), which means some applications will need to be rebuilt...

PeterOGB, 2020-05-28:
See also my problem with installing openGL ES libs (which might be a dupe of this problem).

petiepooo, 2020-05-28:
Would it make sense to have a transition package that just creates symlinks from the locations in /opt/vc to their new places in /usr/lib etc.?  It could be a dependent package for now until the other tools are updated at which point the dependency can be removed.  I would think that would help make a seamless transition while giving the other tools time to respond and be rebuilt.

ghollingworth, 2020-05-28:
One of the reasons we have this beta version is because we know things like this need fixing.  We'd prefer for the functionality not to work on the 64 bit version until everything has been correctly fixed.

Otherwise, you end up with people relying on these links and then complaining when you finally remove them.  Or they stay there forever, which is the worst of both worlds!

A bigger pain now or continuous nagging pain forever?

XECDesign, 2020-05-28:
> There is some package problem when I want to install gnome environment with tasksel

Uninstall pi-package first. You can use `dpkg -r` if `apt` is in a broken state already.

> It seems like libraspberrypi-dev conflicts with a couple of packages.

> See also my problem with installing openGL ES libs (which might be a dupe of this problem). It's a problem with installing the packages not an application building problem (I've not got that far yet!).

Looks like libraspberrypi-dev includes gl headers, which should be excluded on arm64. I'll take them out.

@ghollingworth, any thoughts on moving the old `include/*GL*` to `include/brcm*GL*` for armhf? It would cause some disruption.

> Would it make sense to have a transition package that just creates symlinks from the locations in /opt/vc to their new places in /usr/lib etc.? It could be a dependent package for now until the [...]

I'm not sure what the transition in the armhf repo will look like yet. It's hard to move libs and dev files around without breaking things for people. But it's also required if we want multiarch to work. A package with symlinks may be one way. Another option may be to leave the existing package as is and add new one with a different name and correct paths.

PeterOGB, 2020-05-28:
> Uninstall pi-package first. You can use `dpkg -r` if `apt` is in a broken state already.

Actually needed to unistall libraspberrypi-doc  forst then libraspberrypi-dev but then
apt --fix-broken install worked .

I can now build by openGL ES code :-)    And it works !  Thanks 
PeterO  

PS : I'll close my other report.

XECDesign, 2020-05-28:
Your other report and OP's report are about separate issues. Op had an issue with installing gnome, which is a conflict between gnome-software and pi-package.

PeterOGB, 2020-05-28:
Are you sure ?  Seems like the same problem with same solution.
PeterO

solskogen, 2020-05-28:
This two packages have a conflict, so IMHO this issue should be open until
you can have both packages installed.

XECDesign, 2020-05-28:
> Are you sure ? Seems like the same problem with same solution.

Yeah, I was able to install gnome through tasksel just by removing pi-package-data. Like you said, it doesn't fix the conflict you were seeing and that's because they're between a different set of packages.

> This two packages have a conflict, so IMHO this issue should be open until you can have both packages installed.

I think the plan is to mark pi-package and gnome-software as conflicting with each other so that installing one removes the other, or deprecating pi-package altogether. TBD...

solskogen, 2020-05-28:
The problem is the packages libegl-dev and libraspberrypi-dev.

PeterOGB, 2020-05-28:
> The problem is the packages libegl-dev and libraspberrypi-dev.

But as XECDesign has pointed out, that is NOT the OP's problem that this report is about.  You  have confused thing by mix it up with the libegl/libraspberrypi issue.

PeterO

solskogen, 2020-05-28:
I stand corrected!