raspberrypi/Raspberry-Pi-OS-64bit #10: mmal (buffer header callback not defined)

openbsod, 2020-05-28:
Hi.

Raspberry Pi 4 Model B Rev 1.1
Raspberry Pi Camera Board V2 

raspistill ( and python-picamera ) does not work 

raspistill -v -o foo.jpg

`pi@raspberrypi:~ $ raspistill -v -o foo.jpg

"raspistill" Camera App (commit )

Camera Name imx219
Width 3280, Height 2464, filename foo.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Opening output file foo.jpg
Enabling encoder output port
Starting capture -1
mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
Finished capture -1
mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
Closing down
Close down completed, all components disconnected, disabled and destroyed
`

vcgencmd get_camera

`pi@raspberrypi:~ $ vcgencmd get_camera
supported=1 detected=1
`

v4l2 behaves okej

`
pi@raspberrypi:~ $ v4l2-ctl --set-fmt-video=width=2592,height=1944,pixelformat=3
pi@raspberrypi:~ $ v4l2-ctl --stream-mmap=3 --stream-count=1 --stream-to=somefile.jpg
<
pi@raspberrypi:~ $ file somefile.jpg 
somefile.jpg: JPEG image data, Exif standard: [TIFF image data, big-endian, direntries=9, height=0, model=imx219, xresolution=130, yresolution=138, resolutionunit=2, datetime=1970:01:01 00:00:00, width=0], baseline, precision 8, 2592x1944, components 3
`

https://github.com/raspberrypi/userland/issues/314 for reference

jgan025, 2020-06-01:
Raspberry Pi 4B
Pi Camera v2

I get inconsistent performance from Raspistill and using the PiCamera module to take photos, with the same error;

mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)

The Pi fails to save pictures around 40% of the time. The other 60% it states the error, but still saves the photo correctly.

Hopefully this gets addressed as the camera functionality is core for many Pi users.

cschnepf2, 2020-06-03:
HI,

I'm facing the same issue.

Raspberry 4 B
Pi Camera 1.3

output is the same as @openbsod shared - except I have: 
`
[...]

Camera Name ov5647
Width 2592, Height 1944, filename picture.jpg
Using camera 0, sensor mode 0

[...]

`

lurch, 2020-06-03:
ov5647 simply indicates Camera Module v1 https://www.raspberrypi.org/documentation/hardware/camera/README.md

friendship1, 2020-06-13:
Hi,
with my
Raspberry Pi 4 Model B Rev 1.4,
Raspberry Pi Camera Board V2
I'm facing the same issue, 
output is the completely same as @openbsod shared from the top.

Creating jpg image with "rasptstill" command always fail,
When I try to open the image file, the following error message is displayed:
```Error interpreting JPEG image file (Not a JPEG file: starts with 0x90 0x9f)```

Also, creating image with following python script, displays same error message:
```mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)```

But this time the correct image file is always created.
```python
from picamera import PiCamera
from time import sleep

camera = PiCamera()
camera.rotation = 180

camera.start_preview()
sleep(5)
camera.capture('/home/pi/Desktop/image.jpg')
camera.stop_preview()
```

friendship1, 2020-06-14:
I found the other helpful link.
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=276821&p=1677135&hilit=buffer+header+callback+defined#p1677135
it says:
> _MMAL is not supported on 64bit at the moment_

At the bottom of the page it says:
> __Please post feedback questions on RPiOS64 here: [viewtopic.php?f=63&t=275372](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=275372) and nowhere else

Who manages this Raspberry-Pi-OS-64bit github page?

sheariley, 2022-02-10:
Any news on this?

6by9, 2022-02-10:
Duplicate of https://github.com/raspberrypi/userland/issues/688

> Placeholder so that people can be made aware that MMAL is currently NOT SUPPORTED on 64 bit userland systems