raspberrypi/Raspberry-Pi-OS-64bit #12: pi@raspberrypi:~$ vncserver vncserver: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory

zhangxinmrs, 2020-05-28:
pi@raspberrypi:~$ vncserver
vncserver: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
pi@raspberrypi:~$

XECDesign, 2020-05-28:
That will be resolved either when multi-arch support is added or we get an arm64 build of RealVNC server.

XECDesign, 2020-05-28:
Also, same as https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/3

reliemmp, 2020-06-01:
Jut copy old 32-bit libs to /use/lib/, it works

closed 2020-06-02 by ghollingworth

rilwansmith, 2021-09-20:
The proper solution to this problem is simply to install the arm64 version of the software using the following command.
`apt install realvnc-vnc-server:arm64 -y`
This command will also uninstall the 32bit(armhf) version of the software and restart the vncserver-x11-serviced.service.
Mark or unmark the service to be started at boot time with:
  `systemctl (enable|disable) vncserver-virtuald.service`
Make sure to turn off  auto update of the RealVNC VNC server because RealVNC  website only make available for download the 32bit(armhf) version of the software which is not suitable for Raspberry Pi Os 64bit.

Javadandroid, 2022-11-23:
I solved this in manjaro-arm in these steps:
open terminal and enter:
`sudo ln -fs /opt/vc/lib/libbcm_host.so /usr/lib/libbcm_host.so.0`
`sudo ln -fs /opt/vc/lib/libvcos.so /usr/lib/libvcos.so.0`
`sudo ln -fs /opt/vc/lib/libvchiq_arm.so /usr/lib/libvchiq_arm.so.0`

then:
`systemctl enable  vncserver-virtuald.service`
`systemctl enable vncserver-x11-serviced.service`

`systemctl start vncserver-virtuald.service`
`systemctl start vncserver-x11-serviced.service`
`reboot`

I hope this work for you.