raspberrypi/Raspberry-Pi-OS-64bit #13: Dependency hell installing libsdl2-dev, but fixable by removing libraspberrypi-dev/doc

kaiomatico, 2020-05-28:
Hi, I wanted to compile PPSSPP in 64 bit which requires libsdl2-dev which fails to install because several lib(e)gl(1)-mesa-dev libraries are not allowed to install because of libraspberrypi-dev.  
After then removing libraspberrypi-dev using:  
```
sudo dpkg -P libraspberrypi-doc
sudo dpkg -P libraspberrypi-dev
```
the libsdl2-dev and  lib(e)gl(1)-mesa-dev libraries install properly via apt. PPSSPP then compiles and runs fine in OpenGL mode. So I hope this bug report helps so in the near future we can install libsdl2-dev properly via apt :+1:  Anyway thank you for the awesome 64 bit OS release!!! :)

XECDesign, 2020-05-28:
Sounds like the same root cause as https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/5

pik33, 2020-05-28:
The same error while I tried to sudo apt-get install lazarus.

XECDesign, 2020-05-28:
Fixed packages should be in the repo now

kaiomatico, 2020-05-29:
Great the problem is solved by installing the latest apt updates (did a fresh install to make it super sure) :) I will close this issue now, thank you very much for the fast help

closed 2020-05-29 by kaiomatico