raspberrypi/Raspberry-Pi-OS-64bit #3: VNC does not get enabled trough raspi-config

HDR, 2020-05-28:
Seems like enabling VNC trough raspi-config does not work, on attempting connection after enabling vnc (and restarting) the connection gets "refused by the computer"

spl237, 2020-05-28:
VNC server is not yet available in a version which is compatible with the 64-bit image, so the package has been removed from the image.

MaxLebled, 2020-05-28:
The VNC server won't start due to libbcm_host.so not being found (no such file or directory)

mikebakke, 2020-05-28:
tightvncserver works just fine

claviger-pc, 2020-05-29:
the plethora of issues opened on this shows how many people are using vnc, especially those who use betas.  I did not even try the beta yet because I read these issues first.  But I am taking the opportunity to applaud this move by the RPi dev team and am very excited about a 64bit OS.  Will gladly test the moment realvnc is working, and will subscribe here so that I know.  Cheers!

ling1729, 2020-05-31:
A temporary workaround is to use the x11vnc server instead of the realvnc one. This lacks encryption so it would probably be a bad idea to be port forwarding this, but it's a solution for those that want to access the gui while running a headless setup.
To set it up:
Run `sudo raspi-config` and set a resolution.
Install `x11vnc`:
```
sudo apt-get install x11vnc
sudo x11vnc -storepasswd yourVNCpasswordHERE /etc/x11vnc.pass
sudo nano /etc/systemd/system/x11vnc.service
``` 
Add the following
```
[Unit]
Description="x11vnc"
Requires=display-manager.service
After=display-manager.service

[Service]
ExecStart=/usr/bin/x11vnc -xkb -noxrecord -noxfixes -noxdamage -display :0 -auth guess -rfbauth /etc/x11vnc.pass
ExecStop=/usr/bin/killall x11vnc
Restart=on-failure
Restart-sec=2

[Install]
WantedBy=multi-user.target
```

Start it with 
```
sudo systemctl daemon-reload
sudo systemctl start x11vnc
```
and make it start on boot with 
`sudo systemctl enable x11vnc`

Borillion, 2020-06-04:
Is there a way to get the to set the GL_RENDERER as V3D GL API rather than llvmpipe , with either tightvnc or x11vnc? Its breaking apps that use glfw for example.

spl237, 2020-06-04:
The issues with RealVNC Server on the 64-bit image are being looked into at the moment; we hope to have a version of RealVNC Server which is compatible with 64-bit available in the future. For this reason, we're not going to be looking at alternative VNC solutions for the time being. If you really need VNC, the recommended solution at present is to use the 32-bit image!

MarkDH102, 2020-06-06:
ling1729 
I've tried to get x11vnc running but all my clients (RealVNC, TightVNC) show connection refused. What am I doing wrong?
Fresh installation. (to get rid on NoMachine!)
I can SSH in no problem. I've double checked all the files. I don't get any errors when starting the service.
About to give up with this as it's all mightily frustrating. I may attach a screen etc. tomorrow if I can dig some spares out.

ling1729, 2020-06-06:
See what the output of `sudo systemctl status x11vnc` is. I have found that sometimes a reboot fixes the problem if x11vnc is not starting.

pdfruth, 2020-06-06:
> I've tried to get x11vnc running but all my clients (RealVNC, TightVNC) show connection refused. What am I doing wrong?
Make sure you're using the correct display number (:0 based on linq1729's example above).
If you're configuring the VNC client to use a connection of 'host:1', that could be the cause of your problem.  Try using a connection 'host' instead of 'host:1'.

MarkDH102, 2020-06-06:
I'm really sorry, I had a typo in nofixes which should have been noxfixes! It was only when I was typing this reply that I noticed it. Thank you all for your help on this. So sorry to have wasted your time.

DoraMoed, 2020-08-11:
VNC Server is available??

what the fffffu*k!!!!!!
capture: https://github.com/DoraMoed/js/blob/master/capture_20200811122916487.bmp
 https://github.com/DoraMoed/js/blob/master/capture_20200811123348851.bmp
more: https://github.com/DoraMoed/js/blob/master/vnc-real-more

svenger87, 2020-09-23:
VNC is working again. Just update to the latest version. 
See https://github.com/MichaIng/DietPi/issues/3570#issuecomment-697168935

rmuralidhar, 2020-10-25:
I had an issue with realvnc server installation on ubuntu 20.10 . I could get it to work by the below sequence. 

I assume, anyone who is going to try this has registered with realvnc and has access to download the server .deb package. 

```

cd ~/Downloads
rm -rf ~/Downloads/firmware/
mkdir firmware
cd ~/Downloads/firmware
git init .
git remote add origin https://github.com/raspberrypi/firmware.git
git config core.sparseCheckout true
echo "opt/vc/lib" >> .git/info/sparse-checkout
git pull --depth=1 origin master
sudo mkdir -p opt/vc/lib
sudo cp -r ~/Downloads/firmware/opt/vc/lib/* /opt/vc/lib/

sudo dpkg --install ~/Downloads/VNC-Server-6.7.2-Linux-ARM.deb
sudo apt -f install
sudo dpkg --install ~/Downloads/VNC-Server-6.7.2-Linux-ARM.deb

```

lurch, 2020-10-26:
This is the issue-tracker for the Raspberry Pi OS 64-bit beta, not for Ubuntu on RPi :wink:

tomerrr, 2020-12-02:
this was actually solved but no one here gives a "clear closure" so (at least for me normal update didn't change a thing)
sudo apt remove realvnc-vnc-server 
sudo apt install realvnc-vnc-server:arm64

CaptainMidnight, 2020-12-02:
> this was actually solved but no one here gives a "clear closure" so (at least for me normal update didn't change a thing)
> sudo apt remove realvnc-vnc-server
> sudo apt install realvnc-vnc-server:arm64

As previously stated on the forums - that specific requirement wasn't necessary - standard apt update and upgrades with a possible requirement to re-enable via raspi-config was the only changes to receive the updated working version for the majority of users.

CaptainMidnight, 2020-12-02:
Can an admin please close this issue.

closed 2020-12-02 by JamesH65