raspberrypi/Raspberry-Pi-OS-64bit #4: Missing kernel headers

alexreinert, 2020-05-28:
The image contains the kernel package in version 1.20200527-1
But there is no matching header package available in the APT repository.

satmandu, 2020-05-28:
For what it is worth I have been making functional arm64 header packages for the current firmware rpi kernels for use in ubuntu 64 bit on my RPI4 with this hacky script: https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51

XECDesign, 2020-05-28:
There's no good way to get a headers package without throwing the current kernel package out and doing it the right way from the start. That is a WIP, but won't ship in the near future.

StuartIanNaylor, 2020-05-29:
> There's no good way to get a headers package without throwing the current kernel package out and doing it the right way from the start. That is a WIP, but won't ship in the near future.

I was just trying a custom driver with DKMS and it progressed through armv7 correctly and just stopped on aarch64 because one didn't exist.

It looked like the addition raspberrypi-kernel-headers:aarch would work as expected with raspberrypi-kernel-headers:armhf but is just missing?

tomerrr, 2020-05-29:
> For what it is worth I have been making functional arm64 header packages for the current firmware rpi kernels for use in ubuntu 64 bit on my RPI4 with this hacky script: https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51

i've just tried your script on Raspberry Pi OS which builds deb headers & kernel packages fine , 
but I'm unable to install them - when trying with sudo dpkg -i 

 error setting ownership of './boot/config-5.4.42-v8+': Operation not permitted
when trying to install the headers deb
and
 error setting ownership of '/boot/dtbs.dpkg-new': Operation not permitted
on the linux kernel deb

any idea how to solve this?

XECDesign, 2020-05-29:
> It looked like the addition raspberrypi-kernel-headers:aarch64 would work as expected with raspberrypi-kernel-headers:armhf but is just missing?

Sorry, but no, it's not that simple. It's certainly possible but would add a lot of extra manual steps to the build process involving running things in containers of different architectures in two passes. That will waste me many hours in the future. That time would be better spent upfront fixing the package so that everything just builds as expected in the first place.

pik33, 2020-05-29:
@tomerr: you cannot set ownershgip/permission on /boot because it is fat32 

Maybe unmounting /boot while performing the process and make the false /boot directory will help to finish the process.

tomerrr, 2020-05-29:
@pik33 thank's for the tip! that's exactly what i did (unmounted /boot) and installed the headers deb. then it was just one file to copy to the real /boot directory after remounting it. 
managed to install Wireguard that way

satmandu, 2020-06-01:
@tomerrr @pik33 Sorry, I wrote that script for ubuntu, which has the fat32 partition mounted at /boot/firmware
I should probably look at the headers package on rasbian and see how the scripts are built there...

fatez, 2020-06-02:
> 
> 
> For what it is worth I have been making functional arm64 header packages for the current firmware rpi kernels for use in ubuntu 64 bit on my RPI4 with this hacky script: https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51

Hi,

I am testing your script on a 4GB raspberry 4 with the following operating system : Raspberry Pi OS (64 bit) beta test version : 

> $ uname -ar
> Linux enterprise 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux

I need linux-headers so I can install wireguard when I launch your script comes out with this error:

> exit 0
> dpkg-deb: error: failed to open package info file 'image//DEBIAN/control' for reading: No such file or directory
> mv: cannot stat '/image.deb': No such file or directory

could you tell me what to do?

tomerrr, 2020-06-02:
@fatez 
> 
> > $ uname -ar
> > Linux enterprise 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux
> 
> I need linux-headers so I can install wireguard, but when I launch your script comes out with this error:
> 
> > exit 0
> > dpkg-deb: error: failed to open package info file 'image//DEBIAN/control' for reading: No such file or directory
> > mv: cannot stat '/image.deb': No such file or directory
> 
> could you tell me what to do?

i think you're not using the official raspberry kernel - uname should read

Linux **raspberrypi** 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux

fatez, 2020-06-02:
> @fatez
>
>
> i think you're not using the official raspberry kernel - uname should read
>
> Linux **raspberrypi** 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux

but "raspberry" or "enterprise" are the hostname

tomerrr, 2020-06-02:
oh , sorry. i can run the script again to test what results am i getting now

fatez, 2020-06-02:
you'd be very kind, but as soon as there's a kerel update, I'll be in the same situation.

I need to figure out what's wrong.

satmandu, 2020-06-02:
I'm about to load up the test 64-bit image... so I'll try to get the script fixed then.

satmandu, 2020-06-02:
I had to keep my script from dropping the kernel config file into /boot and trying to set permissions on it.

Once that was commented out, the headers package installs fine.

rand0mbyte, 2020-06-09:
> For what it is worth I have been making functional arm64 header packages for the current firmware rpi kernels for use in ubuntu 64 bit on my RPI4 with this hacky script: https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51

The script works flawlessly and I've got kernel headers for 5.4.44-v8+ working without any issue. +1 Kudos!

pprindeville, 2020-07-09:
> There's no good way to get a headers package without throwing the current kernel package out and doing it the right way from the start. That is a WIP, but won't ship in the near future.

@XECDesign What qualifies as the "near future"? Are we past that yet?

XECDesign, 2020-07-10:
A headers package may or may not show up at some point. I'd like to promise that it will, but something more important always comes along.

pprindeville, 2020-07-13:
> A headers package may or may not show up at some point. I'd like to promise that it will, but something more important always comes along.

Could I try to help out with that? What are the blocking issues?

esbeeb, 2020-07-15:
Wireguard is a must for me, and it's very easy to install in Ubuntu 20.04 64bit for the RPi4.  So that's why this bug means to a lot to me, to be resolved. I'd like for Wireguard to be easy to install, **and not break** every time some kernel security update is released, downloaded, and installed.

esbeeb, 2020-07-20:
> Sorry, but no, it's not that simple. It's certainly possible but would add a lot of extra manual steps to the build process involving running things in containers of different architectures in two [...]

Can you please tell us more about what a proper fix would entail, @XECDesign ?  Can you please break it down for us?

leshniak, 2020-07-26:
> Wireguard is a must for me, and it's very easy to install in Ubuntu 20.04 64bit for the RPi4. So that's why this bug means to a lot to me, to be resolved. I'd like for Wireguard to be easy to install, **and not break** every time some kernel security update is released, downloaded, and installed.

Just make your own build if this is urgent. It's pretty straightforward. I've been doing it since a long time on the RPi itself. This is enough for extracting the headers:
```
export KERNEL_VERSION=`strings arch/arm64/boot/Image | grep 'Linux version' | cut -d' ' -f3`
export KERNEL_HEADERS_PATH="../linux-headers-$KERNEL_VERSION"
export RSYNC_OPTS="-aAXv"

mkdir -pv $KERNEL_HEADERS_PATH
find . -name 'Makefile*' -o -name 'Kconfig*' -o -name '*.pl' | rsync $RSYNC_OPTS --files-from=- . $KERNEL_HEADERS_PATH
find arch/*/include include scripts -type f -o -type l | rsync $RSYNC_OPTS --files-from=- . $KERNEL_HEADERS_PATH
find arch/arm64 -name module.lds -o -name Kbuild.platforms -o -name Platform | rsync $RSYNC_OPTS --files-from=- . $KERNEL_HEADERS_PATH
find $(find arch/arm64 -name include -o -name scripts -type d) -type f | rsync $RSYNC_OPTS --files-from=- . $KERNEL_HEADERS_PATH
find tools/objtool -type f -executable | rsync $RSYNC_OPTS --files-from=- . $KERNEL_HEADERS_PATH
find arch/arm64/include Module.symvers include scripts -type f | rsync $RSYNC_OPTS --files-from=- . $KERNEL_HEADERS_PATH
find scripts/gcc-plugins -name '*.so' -o -name gcc-common.h | rsync $RSYNC_OPTS --files-from=- . $KERNEL_HEADERS_PATH
cp .config $KERNEL_HEADERS_PATH
```
Then move it to the /usr/src and re-link /lib/modules/$KERNEL_VERSION/build to that directory.

cfilipem, 2020-07-26:
Now that we have a 64bit Raspberry Pi OS, the raspberrypi-kernel-headers package for arm64 would be something nice to have. It would allow easier kernel updates when we are running out-of-tree kernel modules.
@XECDesign, can you to increase the priority of fixing this?

T-Birth, 2020-08-02:
For everyone landing on this Page.
I used the following Guide -> https://github.com/notasausage/pi-hole-unbound-wireguard
and had many troubles installing Wireguard on Pi OS64.

With this solution it finally worked:

WATCH OUT ! -> 
Don't use something like "sudo rpi-update" with the current OS64 Beta if u want to get Wireguard working. Had so many troubles with this command.

Don't install raspberrypi-kernel-headers (its mentioned in the Guide above in the category "Setting Up a VPN with WireGuard". Just use this command instead -> 
sudo apt install libelf-dev libmnl-dev build-essential git

Installed Pi OS64 from scratch

Used my little update command -> sudo apt update && sudo apt-get -y upgrade && sudo apt-get -y dist-upgrade && sudo apt-get -y autoremove && sudo apt-get -y autoclean

Used this script to build the kernel headers -> https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51

Installed the builded image with -> sudo dpkg -i  --force-all linux-image-5.4.51-v8+_arm64.deb (some error messages appeard but wayne)

Installed the builded headers with -> sudo dpkg -i  --force-all linux-headers-5.4.51-v8+_arm64.deb

If dependencies are missing use -> sudo apt --fix-broken install

Now Proceed with the Guide above and the installation of Wireguard should work, even on 64.

T-Birth, 2020-08-02:
copy pasted this nice script into this

nano make_deb.sh
chmod +x make_deb.sh
./make_deb.sh

Joulinar, 2020-08-05:
@leshniak 
Sorry for asking this question. I tried to follow your small guide in extracting the headers but I'm already failing on first step to `find` the `Image` file that is used to `grep` the `Linux version`

```
export KERNEL_VERSION=`strings arch/arm64/boot/Image | grep 'Linux version' | cut -d' ' -f3`
```

Pls can you point me into the direction where to find `arch/arm64/boot/Image`

leshniak, 2020-08-05:
@Joulinar you have to compile the desired kernel for ARM64 first. It can be achieved by cross-compiling (with ARM64 toolchain) or compiling in ARM64 chroot environment using the stock ARM64 kernel.

You can check my notes from the build process: https://gist.github.com/leshniak/f2cd4baaebf8b425561dfc5a1ad4ffd4
They are chaotic and messy, but you have some hints.

leshniak, 2020-08-05:
This is my kernel build with headers and WireGuard module (**for RPi 4 only!**):
https://transfer.sh/12DSMa/raspberrypi-kernel_1.20200723-1-build.tar.gz

MD5 checksum:
```
f1311e5fed2cc00357e280f0d49f65fe  raspberrypi-kernel_1.20200723-1-build.tar.gz
```

Extract all files from the root directory perspective. Also, make sure that you have the following entries in the `/boot/config.txt`:
```
# custom kernel
os_prefix=custom/
kernel=kernel8.img

# prefer 64-bit kernel
arm_64bit=1
```

Enjoy!

satmandu, 2020-08-05:
If you just use my script to generate an appropriate kernel headers package, you can install that and also the wireguard-dkms package and not worry about the specifics of compilation... The whole setup becomes much much easier. ;)

satmandu, 2020-08-05:
But this is why I would also strongly suggest that an official headers package be made available with every kernel image release. This is how it works normally for ubuntu and debian kernel releases, since it enables leaving dkms to just do its job in managing external module compilation.

Joulinar, 2020-08-05:
@satmandu 
already tried your script and it was working to get WireGuard up and running. Just like to explore another way 😉 

And as long as we don't have a working headers package, we would need to stay with these workarounds 🙄

satmandu, 2020-08-05:
As I see it ideally (until there is an official set of arm64 headers) you could invoke my script from something in /etc/kernel/postinst.d and also maybe have a wrapper for ```rpi-update``` to do the same.

Joulinar, 2020-08-05:
@leshniak 
thx for guidance. I was able to build my own kernel and extract needed header packages following your guide. But it took a couple of hours compiling 🤣

tomerrr, 2020-08-06:
i used satmandu script a couple of times now and it works great - takes about 5-10 minutes to finish the whole process of installing the kernel headers including unmounting the /boot directory for copying the newly created config* file to /boot

esbeeb, 2020-08-20:
There is talk about moving Raspberry Pi OS to the Linux 5.4 kernel on [this big thread](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&start=425).  Perhaps info can be obtained there, for those who might want to contribute towards an official kernel headers package getting released, which is the original issue here.

heini, 2020-09-20:
/me wondering why this issue is still open, given there's @satmandu's script now. Please use the script to create a kernel headers package and add it to the distribution instead of letting *everyone* go through this process again and again.

satmandu, 2020-09-20:
> /me wondering why this issue is still open, given there's @satmandu's script now. Please use the script to create a kernel headers package and add it to the distribution instead of letting _everyone_ go through this process again and again.

In defense of the RPF package maintainers, I haven't done the work of separating out the headers so that you don't get a conflict if you install the armhf headers onto the same system as the arm64 header package, which might be a problem with the mixed armhf userspace and the optional arm64 kernel if one switches back to a non arm64 kernel.

(But I will say that I use [my script](https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51) to install the headers when I update the kernel on one of my Raspberry PI OS 64-bit systems to get ubuntu's zfs-dkms working - which installs just fine using the RPI kernel and headers with ubuntu's [4700-Fix-DKMS-build-on-arm64-with-PREEMPTION-and-BLK_CGRO](https://git.launchpad.net/ubuntu/+source/zfs-linux/tree/debian/patches/4700-Fix-DKMS-build-on-arm64-with-PREEMPTION-and-BLK_CGRO.patch) patch. )

XECDesign, 2020-10-01:
I've uploaded a reworked headers package today. Let me know how it goes.

The updated package should also allow modules to be cross-compiled from other architectures too.

satmandu, 2020-10-01:
@XECDesign Compiles zfs-dkms from 2.0-rc3 just fine for 5.4.51-v8+.

Any thoughts on an officially supported way to install the headers for a rpi-update kernel? Perhaps some sort of  raspberrypi-kernel-headers-rpi-update-version package that is also marked as providing raspberrypi-kernel-headers?

XECDesign, 2020-10-02:
Thanks for confirming it works.

Not sure how rpi-update headers would work. IIRC, rpi-update has hooks which can run before and after install. So, the package could install hooks which download a headers tarball in preinst and unpack them in postinst. A separate git repo could be set up to host these files, but that would always lag the source repo for the rpi-update kernel. I think it would all need to be handled by the maintainers of rpi-update rather than myself. They could, in theory, push to the headers repo first and then everything would always be in sync. The alternative would be to just say "use rpi-source instead".

@popcornmix, any thoughts?

satmandu, 2020-10-02:
Also, if you just had a hook on your backend which runs when you push kernel updates to generate linux-image and linux-header debs which could be dropped into a folder in the rpi-firmware repos, that could work too. 

rpi-source is just such a big download, and with sd card writing as slow as it can be, it seems suboptimal when you just need sufficient headers to do things like compile out-of-kernel modules.  (My script used to download the entire kernel source, but I fixed that a month back or so.)

popcornmix, 2020-10-06:
@XECDesign I don't think we want to delay rpi-update releases for a headers package to be available, something a small percentage of users are using.

If you do have a script that can generate a headers package then triggering it on a change to https://github.com/Hexxeh/rpi-firmware/blob/master/git_hash might make sense. It would lag rpi-update but only by a couple of hours (I believe it can take an hour for apt packages to reach mirrors, plus the time to build it).

XECDesign, 2020-10-06:
Not quite what I meant. There's no need to delay rpi-update for a headers package. Only a git repo where the headers can be pushed just before the rpi-firmware repo gets updated. It would be a script that fetches the headers from said repo, not an apt package containing the headers.

popcornmix, 2020-10-06:
So this script is something that can be run on buildbot? If so that is possible.

XECDesign, 2020-10-06:
Yup, I'll email you something to generate the appropriate files when I get a chance and if it looks okay, I can have a think about how to make an apt package which will handle the rest. Cheers

XECDesign, 2020-10-06:
Actually, having given it a bit more thought, I don't think that will work. If you cross-compile you end up with some native binaries which aren't usable on target architecture.

satmandu, 2020-10-06:
@XECDesign I haven't dug into it, but the recent headers package on raspi-os is excellent in that it installs headers for all of the architectures (32bit and 64 bit), so that dkms works no matter how you are booted (which is, of course, brilliant). I haven't modified my script to do that (it only installs headers for arm64).

I would suggest generating kernel/rpi-update-release specific linux-headers deb packages because then decluttering by removing old headers from older kernels becomes easier.

(I use the [linux-purge](https://launchpad.net/linux-purge) script for help with decluttering, but I still have to go into /lib/modules and blow away older directories there.)

XECDesign, 2020-10-06:
The new headers package is built natively for armhf and arm64 separately. The same approach wouldn't work for this.

satmandu, 2020-10-06:
@XECDesign Ah. Then 2 packages should work, in theory? Creating the arm64 packages with my script takes less than 15 min on a 4B.

XECDesign, 2020-10-06:
The script that pushes files to rpi-firmware runs on amd64, so without docker or something similar, it's not going to be able to build the headers.

satmandu, 2020-10-06:
@XECDesign Having gotten docker on amd64 to build arm64 headers about a year ago, I'd say that is definitely possible, but the easiest way to deal with this would be to ...
_**puts on snark hat**_ 
get the powers that be to give you some petty cash so you can mosey on down to The Raspberry Pi Store and pick up a couple of those nice 4 Model B devices for use in building a headers packages on demand via a call from the script on the amd64 machine.
**_takes off snark hat and thanks everyone profusely for the work they've put into this already_**

XECDesign, 2020-10-07:
That all sounds great and I know it's all possible, but I have nothing to do with rpi-update in general and there are lots of other things that need to get done that would benefit more people. So, not right now.

tomerrr, 2020-10-25:
i've just updated to kernel 5.4.72-v8 (from official repo) and Wireguard wouldn't compile anymore -
ERROR: "__stack_chk_guard" [/var/lib/dkms/wireguard/1.0.20200506/build/wireguard.ko] undefined!
i've tried removing the Raspberry official headers package and use the above script to create the .deb file from headers source - with same results.
According to this https://rakhesh.com/linux-bsd/wireguard-on-raspberry-pi-os-64-bit/
this has to do with gcc version on buster ..? 
anyway , had to revert back to 5.4.44 with 
rpi-update 65ab2e0
to get Wireguard working again

pprindeville, 2020-10-26:
> rpi-source is just such a big download, and with sd card writing as slow as it can be, it seems suboptimal when you just need sufficient headers to do things like compile out-of-kernel modules. (My script used to download the entire kernel source, but I fixed that a month back or so.)

@satmandu I use a USB 3.0 to mSATA adapter and builds go somewhat faster.

satmandu, 2020-10-26:
@tomerrr  Can't you just run ```sudo apt install -y gcc-10```?

satmandu, 2020-10-26:
@pprindeville Even with just an sd card building isn't so slow now that all the kernel source isn't being packaged. I think the xz compression takes the longest anyways.

XECDesign, 2020-10-26:
@pelwell, @popcornmix could the kernels be compiled with the gcc-8-arm-linux-gnueabihf and gcc-8-aarch64-linux-gnu packages instead, for compatibility with PiOS?

popcornmix, 2020-10-26:
If we are happy that fixes this issue I can do that. @pelwell do you agree?

pelwell, 2020-10-26:
If we can't bump the shipping compiler version then OK.

XECDesign, 2020-10-26:
We possibly could but we shouldn't. Build environments are fragile and I don't know if all the packages in the repo will build with versions of GCC other than what they have already been built with. Changing it could cause issues down the line.

[62 more replies]

closed 2021-03-16 by XECDesign