raspberrypi/Raspberry-Pi-OS-64bit #5: Errors installing openGL ES development packages

PeterOGB, 2020-05-28:
Trying to install pakages needed for  open GL ES development...  (libgles2-mesa-dev)  produces these errors.

npacking libgl-dev:arm64 (1.3.0-7~bpo10+1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-Omh6jw/64-libgl-dev_1.3.0-7~bpo10+1_arm64.deb (--unpack):
 trying to overwrite '/usr/include/KHR/khrplatform.h', which is also in package libraspberrypi-dev 2+git20200520~143707+2fe4ca3-1

Unpacking libegl-dev:arm64 (1.3.0-7~bpo10+1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-Omh6jw/65-libegl-dev_1.3.0-7~bpo10+1_arm64.deb (--unpack):
 trying to overwrite '/usr/include/EGL/egl.h', which is also in package libraspberrypi-dev 2+git20200520~143707+2fe4ca3-1

dpkg: error processing archive /tmp/apt-dpkg-install-Omh6jw/72-libgles-dev_1.3.0-7~bpo10+1_arm64.deb (--unpack):
 trying to overwrite '/usr/include/GLES/gl.h', which is also in package libraspberrypi-dev 2+git20200520~143707+2fe4ca3-1

Errors were encountered while processing:
 /tmp/apt-dpkg-install-Omh6jw/64-libgl-dev_1.3.0-7~bpo10+1_arm64.deb
 /tmp/apt-dpkg-install-Omh6jw/65-libegl-dev_1.3.0-7~bpo10+1_arm64.deb
 /tmp/apt-dpkg-install-Omh6jw/72-libgles-dev_1.3.0-7~bpo10+1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

PeterOGB, 2020-05-28:
FIxed with advice from XECdesign in "there is some package problem when I want to install gnome environment with tasksel "

closed 2020-05-28 by PeterOGB

XECDesign, 2020-05-28:
Fixed packages should be in the repo now

PeterOGB, 2020-05-29:
Can confirm this is now fixed in the repo.  Thanks for quick resolution.

geerlingguy, 2020-06-09:
I'm trying to do this today and getting:

```
Unpacking libnautilus-extension-dev:arm64 (3.30.5-2) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-CBkmxo/49-libgl-dev_1.3.0-7~bpo10+1_arm64.deb
 /tmp/apt-dpkg-install-CBkmxo/50-libegl-dev_1.3.0-7~bpo10+1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

(While trying to install `libnautilus-extension-dev` as a dependency of Dropbox.)

XECDesign, 2020-06-11:
You still had the old libraspberrypi-dev package. I don't think you would've seen the error if you updated the system first.

tomh05, 2020-10-12:
I found the footnote in @geerlingguy 's post worked for me:
```
sudo apt-get -o Dpkg::Options::="--force-overwrite" install libgles2-mesa-dev
```