raspberrypi/Raspberry-Pi-OS-64bit #7: Not able to compile custom kernel

rbollampally, 2020-05-28:
When compiling custom Kernel on RPI4 - 4GB running 64bit OS, I get the following error:

```
pi@raspberrypi:~/linux $ KERNEL=kernel8
pi@raspberrypi:~/linux $ make bcm2711_defconfig
#
# configuration written to .config
#
pi@raspberrypi:~/linux $ make -j4 zImage modules dtbs
scripts/kconfig/conf  --syncconfig Kconfig
make: *** No rule to make target 'zImage'.  Stop.
pi@raspberrypi:~/linux $

```

rbollampally, 2020-05-28:
Ignore. It should be "Image" not "zImage" for 64bit

closed 2020-05-28 by rbollampally

lurch, 2020-05-29:
ping @JamesH65 @pelwell ?

JamesH65, 2020-05-29:
I guess we are going to need some adjustments to the kernel building pages to cope with 64 bit compiling. Is that likely to be a moving target at the moment, or could it be done now?

pelwell, 2020-05-29:
I can't see the build instructions changing much. Only the DTB installation could do with a bit of work.