raspberrypi/Raspberry-Pi-OS-64bit #8: Touchscreen off by half resolution using the official 7" 800x480 TFT

xranby, 2020-05-28:
The touchscreen places the pointer at aproximately x*2 and y*2 when the the official 7" 800x480 TFT
[image]

It is almost like the Raspberry-Pi-OS-64bit expects the screen to be 1600x960, 
when can we get the new screen? ;)

AdamorfuX, 2020-05-29:
Hi!

Not sure if it works but you can try installing xinput-calibrator with sudo apt-get install xinput-calibrator

The go to menu - preferences - Calibrate Touchscreen

I hope it helps

xranby, 2020-05-29:
i tested xinput-calibrator but it appears that the entire screen is too large by default, thus i cant touch all calibration points
[image]

xranby, 2020-05-29:
The issue:
It appears that the composite signal is enabled by default
it makes the screen configuration look like this:
[image]

Workaround:
re arrange the composite screen to stay below the DSI-1 screen workaround the issue.
[image]

xranby, 2020-06-01:
Workaround/fix:
Add `enable_tvout=0` to config.txt

ghollingworth, 2020-06-02:
This is the same on the 32bit kernel, so you should report this in the firmware repo instead.

Currently if no HDMI is detected the firmware will set up the composite output.  With KMS this now means it has both displays and adds them.  So you'd either have to disable the tvout (as above) or we'd have to change default behaviour...

This issue should be fixed once we fully move to the 'full KMS' solution and fix both the DSI driver and tvout drivers in the kernel so that you can enable / configure the interfaces in linux.  For the moment though we need to default between HDMI and tvout because we don't know at that time whether a DSI display is available.

I'll leave this open, as a reminder, but it should really not be here.

lurch, 2020-06-02:
Looks (at least to me) like the problem is that the touchscreen-input is being "scaled" to cover both the DSI and composite display outputs. Does that suggest that a different "fix" would be to modify the touchscreen driver so that its input-events only cover the DSI display, not all displays? :shrug:

xranby, 2020-06-02:
@lurch raises a valid point, if i start the system with HDMI attached and DSI enabled then the pointer is misaligned in the same way. Thus the proposed fix to make the touchscreen driver send input events that only cover the DSI display, if that is possible, sounds like a good solution.
[image]

6by9, 2020-10-23:
xinput is the command you're after
https://www.raspberrypi.org/forums/viewtopic.php?p=1724631#p1724631

```xinput --list``` lists all the input devices, then 
```xinput map-to-output 10 DSI-1``` where 10 is the appropriate number from that list.

xranby, 2021-01-08:
so the solution to lock the touchscreen to the TFT display would be to add an Xsession containing: ?
`xinput map-to-output "raspberrypi-ts" DSI-1`