raspberrypi/Raspberry-Pi-OS-64bit #9: Make a lite version available

NilsIrl, 2020-05-28:
Hi, would it be possible to make a "lite" version available? Like it is for the 32bit releases.

MichaIng, 2020-05-28:
Totally agree. However for now I think its easier to have a single image only for the first wave of bug reports to test and verify against.

hsxsix, 2020-05-29:
Agree it, just wait patiently...

BertalanD, 2020-06-08:
I've been using 64-bit Debian on my 3B+ for 2 months. It works flawlessly with Pi-Hole and Unbound, even with a 30 day uptime (I restarted it once due to a kernel update). Kernel is mainline 4.19  and does everything ~Raspbian~ *Raspberry Pi OS* does except the `vcgencmd` userland stuff. The new instructions introduced in `armv8` do make it faster, so I find it nice even though 1 GB of RAM does not warrant it. Currently this is as close as I can get to a 64-bit lite ~Raspbian~ *Raspberry Pi OS*. 

The only reason keeping me from getting a Pi 4 is that Debian does not support it.

MichaIng, 2020-06-08:
> The only reason keeping me from getting a Pi 4 is that Debian does not support it, but maybe a 64-bit Raspberry Pi OS release will change my mind.

Since when Debian does not support it? Of course you need the RPi kernel (add the archive.raspberrypi.org/debian repository to your sources, install via debootstrap or chroot) aside from this Debian is working pretty fine on all RPi2 and up models. And 64-bit _Raspberry Pi OS_ IS in fact Debian with the RPi kernel repository and its kernel/bootloader/dtb pre-installed, nothing else.

lurch, 2020-06-08:
> And 64-bit _Raspberry Pi OS_ IS in fact Debian with the RPi kernel repository and its kernel/bootloader/dtb pre-installed, nothing else.

And the desktop tweaks, and various pre-installed applications :wink:  I suspect some purists are more pure than others? :thinking: :laughing:

MichaIng, 2020-06-08:
> And the desktop tweaks, and various pre-installed applications 😉 I suspect some purists are more pure than others? 🤔 😆

Indeed 😅, a lite version will be without the desktop tweaks and, more importantly, all those pre-installed packages are not part of Raspbian but from the archive.raspberrypi.org/debian repository, hence can be installed on Debian as well. Although, there are some special binary builds as well which are armv6hf, hence not compatible with Debian, but not relevant for arm64, where most of those are still missing, AFAIK, hostapd, Kodi and such.

lurch, 2020-06-08:
Looks like https://packages.debian.org/search?keywords=hostapd&searchon=names&suite=stable&section=all and https://packages.debian.org/search?keywords=kodi&searchon=names&suite=stable&section=all both exist in arm64 for Debian? Of course the "pure Debian" version of kodi may not have the full RPi-specific hardware acceleration.

MichaIng, 2020-06-08:
From what I know does the Debian version of hostapd lack support for some Realtek WiFi adapters (RTL8188C*)  and
> "pure Debian" version of kodi may not have the full RPi-specific hardware acceleration

exactly 👍.

ifjq217, 2020-07-19:
For the impatient... the arm64 branch of py-gen makes it possible to build the 'lite' 64-bit image. https://github.com/RPi-Distro/pi-gen/tree/arm64

It's working well for me.

XECDesign, 2020-08-25:
https://downloads.raspberrypi.org/raspios_lite_arm64/images/

lurch, 2020-11-15:
@ghollingworth This can be closed.

closed 2020-11-15 by ghollingworth