raspberrypi/Raspberry-Pi-OS-64bit #14: Broken libgl-dev

Borillion, 2020-05-29:
So I ended up with a broken version of libgl-dev when I tried to install it and vncserver. 

pi@raspberrypi:~ $ sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian-security buster/updates InRelease
Hit:3 http://deb.debian.org/debian buster-updates InRelease
Hit:4 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done                             
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libgl1-mesa-dev : Depends: libgl-dev
 mesa-common-dev : Depends: libgl-dev
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

pi@raspberrypi:~ $ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  libgl-dev
The following NEW packages will be installed:
  libgl-dev
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
83 not fully installed or removed.
Need to get 100 kB of archives.
After this operation, 1,389 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian buster/main arm64 libgl-dev arm64 1.3.0-7~bpo10+1 [100 kB]
Fetched 100 kB in 1s (148 kB/s)   
(Reading database ... 97636 files and directories currently installed.)
Preparing to unpack .../libgl-dev_1.3.0-7~bpo10+1_arm64.deb ...
Unpacking libgl-dev:arm64 (1.3.0-7~bpo10+1) ...
dpkg: error processing archive /var/cache/apt/archives/libgl-dev_1.3.0-7~bpo10+1_arm64.deb (--unpack):
 trying to overwrite '/usr/include/KHR/khrplatform.h', which is also in package libraspberrypi-dev 2+git20200520~143707+2fe4ca3-1
Errors were encountered while processing:
 /var/cache/apt/archives/libgl-dev_1.3.0-7~bpo10+1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

PeterOGB, 2020-05-29:
There are three or so reports about this already.  Maybe read them and find the solution !

pelwell, 2020-05-29:
But there are so many issues to read the through - we're into double figures already.

PeterOGB, 2020-05-29:
WOW! such lazyness ! Just reading the titles will get you to the (already fixed in the repo) solution .

Borillion, 2020-05-29:
So assume that this is the solution? 

https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/13

Issue for me is I don't know what libraspberrypi-dev does, and what side effects removing it might have. So when not certain I believe it's better to ask. Personally prefer people not comment if they don't have anything to helpful or positive to contribute.

PeterOGB, 2020-05-29:
The solution is to update your system to use the latest packages in the repo as they have been fixed.

Personally I would prefer people to look at the other issues before filling up the page with duplicates of already reported (and solved) issues.

Borillion, 2020-05-29:
I appreciate your help regardless. So apt-get update, upgrade & dist-upgrade clearly didn't get the update, how do I update to use the correct repo?

ghollingworth, 2020-05-29:
You shouldn't use apt-get anymore, you should use 
sudo apt update
sudo apt  full-upgrade

closed 2020-05-29 by Borillion