raspberrypi/Raspberry-Pi-OS-64bit #16: Raspberry Pi 4 Model  Raspberry Pi OS (64 bit )

marellanrao, 2020-05-30:
I have tried Raspberry Pi OS (64bit) for Raspberry Pi 4 Model B through SD card.Its ok.But after copying to WD SSD(240GB) os is loading but internet issue is coming.
Anybody can help me to solve the issue.

lurch, 2020-05-30:
I'm afraid you'll need to provide **much** more info than that - from your brief description it's impossible to tell what your problem is, let alone what might be causing it...

mehulboricha, 2020-05-30:
I did encounter the internet issue. In my case, it was a strange DNS problem. However, this fixed it:

https://raspberrypi.stackexchange.com/questions/64556/problem-with-dns

fishd72, 2020-06-01:
Can confirm I'm running the 64-bit beta on a Raspberry Pi 4 and have used wifi and ethernet, both from an SD card and booting from a USB connected SSD. No issues found with internet/network connectivity so far.

marellanrao, 2020-06-02:
First of all I have installed Raspberry Pi OS(64 bit) on 32GB memory card
and used in my Raspberry Pi 4 Model B(4GB).
The software runs smoothly.
After that I have copied the os from memory card to WD SSD(240 GB).I have
removed the memory card and and started the Pi 4, it boosted without Any
problem.
But Ethernet is not working.
Is there any modifications to be done to run Internet.
I am amature in software.

Narasimha Rao Marella

mehulboricha, 2020-06-02:
Exactly the same thing happened to me. What I did was to mention DNS servers manually in resolve.conf file. Try the link that I have mentioned earlier.

marellanrao, 2020-06-02:
Thank you very much and I will try it.

[image: Narasimharao Marella on about.me]

Narasimharao Marella
about.me/NarasimharaoMarella
  <http://about.me/NarasimharaoMarella>

closed 2020-06-02 by ghollingworth