raspberrypi/Raspberry-Pi-OS-64bit #18: Does it work on Raspberry PI 1B

ameer1234567890, 2020-05-30:
Not an issue, but a question. Can I run this on a Raspberry Pi 1B?

Just asking before I try.

theAkito, 2020-05-30:
You need a 64-bit CPU, so it cannot work on a Raspberry Pi older than the 3rd generation.

lurch, 2020-05-30:
https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370 clearly says "Note, the 64bit OS is only install-able on the Pi 3 and Pi 4 devices".

ameer1234567890, 2020-05-30:
This would essentially mean that the Raspberry Pi foundation has to maintain a 32 bit OS for older Raspberry Pi's while maintaining a 64 bit OS for newer Pi's, unless dropping support for older Pi's is decided.

pelwell, 2020-05-30:
There is no end date for 32-bit support. Your question has been answered, so please close the issue.

theAkito, 2020-05-30:
> This would essentially mean that the Raspberry Pi foundation has to maintain a 32 bit OS for older Raspberry Pi's while maintaining a 64 bit OS for newer Pi's, unless dropping support for older Pi's is decided.

Look at Ubuntu as an example. They dropped 32-bit and now people are using alternatives. The same would happen with Raspberry Pi, if the demand would be as high. So don't worry about your ancient machine.

Xrunes, 2020-05-31:
No, you can't. 64bit system only support pi3 and pi4. But don't worry, the raspberry pi official will not stop the maintenance of 32bit system in a short time, because 64bit system still in beta stage.

closed 2020-05-31 by ameer1234567890