raspberrypi/Raspberry-Pi-OS-64bit #19: Sysbench not available on apt

Notupus, 2020-05-30:
APT is unable to get the Sysbench package seems like some repository problem

XECDesign, 2020-05-30:
Sysbench doesn't exist in Debian Buster - https://packages.debian.org/search?suite=buster&searchon=names&keywords=sysbench

Notupus, 2020-05-30:
I mean try.

Notupus, 2020-05-30:
It worked on 32 bit

XECDesign, 2020-05-30:
Because Raspbian doesn't clear old packages out when there's a new release. It probably has the old Stretch package in Buster.

Notupus, 2020-05-30:
Ok, it seems I need a replacement

lurch, 2020-05-30:
It **is** in buster-backports though: https://packages.debian.org/buster-backports/sysbench

Maybe you'll be able to download the arm64 deb and install it manually? Note that I've not tried that myself though, so it may or may not work!

closed 2020-06-02 by ghollingworth

raginigupta6, 2020-11-11:
How can I install sysbench on Rasbpian (64bit)

lurch, 2020-11-12:
https://backports.debian.org/Instructions/ might help, but that's obviously entirely unsupported! (I've not tried it myself)