raspberrypi/Raspberry-Pi-OS-64bit #21: OpenGL Driver with Fake KMS

Borillion, 2020-05-31:
Looks like the OpenGL driver, "G2 GL (Fake KMS) OpenGL desktop driver with fake KMS" isn't working. Is there something I missed?

glxinfo
name of display: :1.0
Error: couldn't find RGB GLX visual or fbconfig

glxgears
Error: couldn't get an RGB, Double-buffered visual

XECDesign, 2020-05-31:
Can't reproduce.

ghollingworth, 2020-06-02:
Can you confirm what you are doing here to make this happen?

Borillion, 2020-06-04:
Not sure if this coherent enough, let me know. So this issue here seems to do with how the opengl driver is loaded at least in a 32 bit OS that does work. It appears there is a difference between running vncserver from the terminal and using raspi-config to install it.

Rasberry Pi OS 05/27/2020 32 Bit just running vncserver on the terminal over ssh gets me this.
pi@raspberrypi:~ $ glxgears -info
GL_RENDERER = llvmpipe (LLVM 9.0.1, 128 bits)
GL_VERSION = 3.1 Mesa 19.3.2
GL_VENDOR = VMware, Inc.
I also see the same on the 64 bit version of the os, however I manually installed tightvnc server.

Output thats working for me, after using raspi-config to enable vncserver and GL with fake KMS. 
GL_RENDERER = V3D 4.2
GL_VERSION = 2.1 Mesa 19.3.2
GL_VENDOR = Broadcom

That kinda links back to what I posted here, https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/3#issuecomment-638565731

XECDesign, 2020-06-04:
Sounds like you're running a virtual vnc display rather than a real one. If you use something like x11vnc, it should use v3d.

Borillion, 2020-06-05:
That was exactly it. Thanks. Good learning experience. I didn't know about physical and virtual VNC sessions :)

closed 2020-06-05 by Borillion

Predictfutur, 2023-06-18:
Title: Issue with installing Intel RealSense SDK on Raspberry Pi

Description:
I am encountering difficulties while trying to install the Intel RealSense SDK on my Raspberry Pi 3B model with an Intel RealSense D435i camera. I have followed the recommended steps provided in the official documentation, but I'm facing issues with the repository setup and package installation.

I have tried adding the Intel RealSense repository to the apt sources list, but I'm receiving errors like "403 Forbidden" and "The repository is not signed." This prevents me from installing the necessary packages for the SDK.

I have also attempted to build the SDK from source, but encountered various issues and dependencies that were difficult to resolve.

I am seeking assistance to successfully install the Intel RealSense SDK on my Raspberry Pi device. Are there any alternative methods or specific steps that I should follow?

Any guidance or suggestions would be greatly appreciated.

System Details:
- Raspberry Pi Model: 3B
- Intel RealSense Camera Model: D435i

Thank you!