raspberrypi/Raspberry-Pi-OS-64bit #22: CONFIG_USB_FUNCTIONFS disabled in kernel

burtyb, 2020-05-31:
Could CONFIG_USB_FUNCTIONFS be enabled in the kernel please (I've previously been using it to setup composite USB Gadgets).

pelwell, 2020-06-02:
> (I've previously been using it to setup composite USB Gadgets)

Was this in custom kernel builds? I can't find any evidence that we've ever defined it.

maxnet, 2020-06-02:
>Could CONFIG_USB_FUNCTIONFS be enabled in the kernel please (I've previously been using it to 
>setup composite USB Gadgets).

You can use the configfs interface provided through CONFIG_USB_CONFIGFS and friends (CONFIG_USB_CONFIGFS_*) instead, to setup composite USB gadgets.

burtyb, 2020-06-02:
Yikes, I'm sorry you're right I'd mixed up two projects.

closed 2020-06-02 by burtyb