raspberrypi/Raspberry-Pi-OS-64bit #25: Ubuntu server 64 bit can access only 4gb not 8gb ram

samlearner-prog, 2020-06-01:
I have researched ubuntu forums as well, but most discussions point towards a firmware issue.  pls help check.

kencormack, 2020-06-01:
This forum is for reporting problems with the "Raspberry Pi OS" 64-bit beta OS, not Ubuntu Server 64-bit

satmandu, 2020-06-01:
@samlearner-prog I don't have my 8Gb pi yet but can you see if the official raspberry pi foundation kernels work properly with the 8gb rpi4b?

Try using this script to build a kernel package deb that you can install on ubuntu arm64 server:

https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51

kencormack, 2020-06-01:
The official "Raspbian" Buster (32-bit) image should work just fine.  I'm not a developer for this 64-bit effort.  I was just pointing out that your thread seems to suggest you wanted to talk about Ubuntu Server, and not "Raspberry Pi OS" (the name for the Pi's beta 64-bit kernel.)  Just wanted to make sure you were aware this forum isn't for discussing anything but the beta.

samlearner-prog, 2020-06-01:
@kencormack - i realised just a little too late.  apologies. 
@ satmandu  - many thanks for the pointer. 
i will close

closed 2020-06-01 by samlearner-prog