raspberrypi/Raspberry-Pi-OS-64bit #28: kdevelop failes to install

camccar, 2020-06-01:
Installing **kdevelop** works in the 32bit Rasbian, but when trying to install kdevelop in `Raspberry-PI-OS-64bit` I get the following errors,

```
pi@raspberrypi:~ $ sudo apt-get install kdevelop
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kdevelop : Depends: kdevelop53-libs (= 4:5.3.1-3) but it is not going to be installed
            Depends: kinit but it is not going to be installed
            Depends: kio but it is not going to be installed
            Depends: libkasten3controllers3 (>= 5:0.25.3~) but it is not going to be installed
            Depends: libkasten3core3 (>= 5:0.25.3~) but it is not going to be installed
            Depends: libkasten3okteta1controllers1abi1 (>= 5:0.25.3) but it is not going to be installed
            Depends: libkasten3okteta1core1 (>= 5:0.25.3~) but it is not going to be installed
            Depends: libkasten3okteta1gui1 (>= 5:0.25.3~) but it is not going to be installed
            Depends: libkf5declarative5 (>= 5.45.0) but it is not going to be installed
            Depends: libkf5newstuff5 (>= 5.27.0) but it is not going to be installed
            Depends: libkf5parts5 (>= 4.96.0) but it is not going to be installed
            Depends: libkf5purpose-bin but it is not going to be installed
            Depends: libkf5purpose5 (>= 1.1) but it is not going to be installed
            Depends: libkf5texteditor5 (>= 5.15.0) but it is not going to be installed
            Depends: libkomparediff2-5 (>= 4:14.12.50) but it is not going to be installed
            Depends: libprocessui7 (>= 5.2.50+git) but it is not going to be installed
            Recommends: kapptemplate but it is not going to be installed
            Recommends: kio-extras but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

camccar, 2020-06-01:
Looks like `sudo apt-get install phonon4qt5` fixed it for me. ¯\_(ツ)_/¯

XECDesign, 2020-06-01:
It's hard to tell what went wrong there. I tried it on my system and everything installed without any issues. Did you install any armhf libraries that could've caused conflicts? Did installing phonon4qt5 remove anything? You can check by looking in /var/log/apt/history.log.

camccar, 2020-06-01:
Does not look like it. if I search that log file for `phoneon4qt5` I get,

```
Start-Date: 2020-06-01  16:17:52
Commandline: apt-get install phonon4qt5
Requested-By: pi (1000)
Install: phonon4qt5:arm64 (4:4.10.2-1), phonon4qt5-backend-gstreamer:arm64 (4:4.9.0-1, automatic), libqt5x11extras5:arm64 (5.11.3-2, automatic), libpulse-mainloop-glib0:arm64 (12.2-4+deb10u1, automatic), phonon\
-backend-gstreamer-common:arm64 (4:4.9.0-1, automatic), libphonon4qt5-4:arm64 (4:4.10.2-1, automatic), libqt5opengl5:arm64 (5.11.3+dfsg1-1+deb10u3, automatic)
End-Date: 2020-06-01  16:17:54

```
Next entry is the install kdevelop command.

XECDesign, 2020-06-01:
That's a new one for me. Not sure what happened there or how to reproduce it.

camccar, 2020-06-02:
Closing because it was probably the result of a random package I installed

closed 2020-06-02 by camccar