raspberrypi/Raspberry-Pi-OS-64bit #29: VLC Not Available to install

camccar, 2020-06-01:
VLC is not available. When I try to install it,

```
pi@raspberrypi:~/projects $ sudo apt-get install vlc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 vlc : Depends: vlc-plugin-base (= 3.0.10-0+deb10u1) but it is not going to be installed
       Recommends: vlc-l10n (= 3.0.10-0+deb10u1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

Tested on a rpi4 8gb booting on usb with the may28th firmware.

XECDesign, 2020-06-01:
That will be fixed, but as a workaround you should be able to run `sudo apt install vlc vlc-data=3.0.10-0+deb10u1` to install it.

camccar, 2020-06-01:
Thank you that worked. I can listen to my music collection while doing work now.