raspberrypi/Raspberry-Pi-OS-64bit #30: Rasptstill, picamera not working

jgan025, 2020-06-01:
Raspberyr Pi 4B
Pi Camera v2

Trying to take pictures with Raspistill command (and picamera module) results in the following error;

  _mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)_

Sometimes the picture it taken and stored correctly, however often the resulting file size is very small, and attempting to open the image generates the following error;

  _Not a JPEG file: starts with 0x00 0x00_

Hopefully we find a solution soon as capturing pictures on a Pi is pretty fundamental to most projects.

Thanks

lurch, 2020-06-01:
Duplicate of #10 ?

jgan025, 2020-06-01:
Yup, my bad. Was including my issue with that post but also wanted to open a seaparate one for just still images. Although I guess the problems are one and the same