raspberrypi/Raspberry-Pi-OS-64bit #31: No audio from snap applications

Covin90, 2020-06-01:
After installing retroarch or whatever snapd application I get this error and I don't receive any audio at all. 

`pi@raspberrypi:~ $ sudo snap install retroarch
2020-06-02T01:15:13+02:00 INFO snap "retroarch" has bad plugs or slots: audio-playback (unknown interface "audio-playback")
retroarch 1.8.7 from hunterk (libretro) installed
pi@raspberrypi:~ $ retroarch
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
sh: 1: xdg-screensaver: not found
[CDROM] No sg devices found and sg kernel module is not loaded.
ALSA lib conf.c:3916:(snd_config_update_r) Cannot access file /usr/share/alsa/alsa.conf
ALSA lib seq.c:935:(snd_seq_open_noupdate) Unknown SEQ default
Failed to open /sys/class/power_supply/hidpp_battery_0/capacity: No such file or directory
`
Any solution for this? 

Thanks

XECDesign, 2020-06-02:
It sounds like this has more to do with snap and that particular snap package than the 64 bit image itself. I'm not sure that this is the best place to get support for snap packages.

ghollingworth, 2020-06-02:
It would be interesting to see if you install pulseaudio