raspberrypi/Raspberry-Pi-OS-64bit #32: Can't compile with wiringpi

MishterKirby, 2020-06-02:
This is the output I get after compiling a tutorial from a Freenove kit:

`/usr/bin/ld: skipping incompatible /usr/lib/gcc/aarch64-linux-gnu/8/../../../../lib/libwiringPi.so when searching for -lwiringPi
/usr/bin/ld: skipping incompatible /usr/lib/../lib/libwiringPi.so when searching for -lwiringPi
/usr/bin/ld: skipping incompatible /usr/lib/gcc/aarch64-linux-gnu/8/../../../libwiringPi.so when searching for -lwiringPi
/usr/bin/ld: skipping incompatible //usr/lib/libwiringPi.so when searching for -lwiringPi
/usr/bin/ld: cannot find -lwiringPi
collect2: error: ld returned 1 exit status`

XECDesign, 2020-06-02:
There's no aarch64 build for wiringpi in the repo right now. There's an armhf toolchain in the repo that should work, although I haven't tried it with wiringpi myself.

I'm not sure what level of support we can offer for wiringpi moving forward, so please consider switching to an alternative.

http://wiringpi.com/wiringpi-deprecated/

closed 2020-06-02 by ghollingworth

wujianzhang1237, 2020-06-11:
> 现在在回购中没有用于连接pi的aarch64构建。回购中有一个应该可以使用的armhf工具链，尽管我自己没有用wirepi尝试过。
> 
> 我不确定我们可以为接线提供什么样的支持，因此请考虑改用其他方法。
> 
> http://wiringpi.com/wiringpi-deprecated/

This tool chain seems to be unable to compile the wiringPI program, too difficult, may I ask when the official can optimize these as soon as possible

wujianzhang1237, 2020-06-11:
> There's no aarch64 build for wiringpi in the repo right now. There's an armhf toolchain in the repo that should work, although I haven't tried it with wiringpi myself.
>
> I'm not sure what level of [...]

Now, this chain of tools doesn't look like wiring PI anymore, it's too hard. When can we optimize this as soon as possible

ghollingworth, 2020-06-11:
Not sure what you mean by 'optimize this.'  The developer of wiringpi has dropped support for it, as such it will not be supported from this point on.  Neither on 32 or 64 bit Raspbian.

I can only suggest you look for an alternative library to accomplish whatever it is you need to do.  Probably the right one is libgpiod

https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/README

watershade, 2020-09-24:
I think github.com/wiringpi/wiringpi is a great alternative. I have test it. It is worked well