raspberrypi/Raspberry-Pi-OS-64bit #33: Wolfram Mathematica fails to install

nwalton2502, 2020-06-02:
sudo apt install wolfram-engine returns

Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 wolfram-engine:armhf : Depends: oracle-java7-jdk:armhf but it is not installable or
                                 oracle-java8-jdk:armhf but it is not installable or
                                 openjdk-11-jdk:armhf but it is not going to be installed
                        Depends: libssl1.0.2:armhf but it is not installable
                        Depends: liboauth0:armhf but it is not going to be installed
                        Depends: libgfortran3:armhf but it is not installable

Java is installed 

java -version
openjdk version "11.0.7" 2020-04-14
OpenJDK Runtime Environment (build 11.0.7+10-post-Debian-3deb10u1)
OpenJDK 64-Bit Server VM (build 11.0.7+10-post-Debian-3deb10u1, mixed mode)

Fortran installs by hand but with libgfortran5 instead of libgfortran3

/var/lib/dpkg/info/libgfortran5:arm64.triggers
/var/lib/dpkg/info/libgfortran5:arm64.list
/var/lib/dpkg/info/libgfortran5:arm64.md5sums
/var/lib/dpkg/info/libgfortran-8-dev:arm64.list
/var/lib/dpkg/info/libgfortran5:arm64.symbols
/var/lib/dpkg/info/libgfortran5:arm64.shlibs
/var/lib/dpkg/info/libgfortran-8-dev:arm64.md5sums
/var/cache/apt/archives/libgfortran-8-dev_8.3.0-6_arm64.deb
/usr/lib/aarch64-linux-gnu/libgfortran.so.5.0.0
/usr/lib/aarch64-linux-gnu/libgfortran.so.5
/usr/lib/gcc/aarch64-linux-gnu/8/libgfortran.a
/usr/lib/gcc/aarch64-linux-gnu/8/libgfortran.spec
/usr/lib/gcc/aarch64-linux-gnu/8/libgfortran.so

ghollingworth, 2020-06-02:
Added to known issues list, we will have to wait for Wolfram here

vb0, 2020-07-03:
I also tried the free [Wolfram Engine](https://www.wolfram.com/engine/free-download-thank-you/?operatingSystem=LINUX) - that actually is less than what we used to get on the Pi with the wolfram-engine on the Pi (which included nicely also Mathematica, the notebook interface, although kind of too slow to be really useful, at least for me). Anyway that didn't work as it's trying to install a amd64 deb. In turn Sagemath build and works well, [for the first time on the Pi](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/43#issuecomment-653385510) (at least for me).

danielktdoranie, 2020-09-01:
> sudo apt install wolfram-engine returns
>
> Reading package lists...
> Building dependency tree...
> Reading state information...
> Some packages could not be installed. This may mean that you have
> requested [...]

Same.

ilian8, 2020-09-09:
While still armhf, [wolfram-engine_12.1.1+2020081901_armhf.deb](http://archive.raspberrypi.org/debian/pool/main/w/wolfram-engine/wolfram-engine_12.1.1+2020081901_armhf.deb) should install fine

ilian8, 2023-03-08:
Starting with version 13.2.1, e.g. [wolfram-engine_13.2.1+20230227524_arm64.deb](https://archive.raspberrypi.org/debian/pool/main/w/wolfram-engine/wolfram-engine_13.2.1+20230227524_arm64.deb), Mathematica is 64-bit.