raspberrypi/Raspberry-Pi-OS-64bit #34: Install language failed

ss8688, 2020-06-02:
Failed to install Chinese language when initialize

ghollingworth, 2020-06-03:
Can you provide any instructions to follow to reproduce this?

ss8688, 2020-06-03:
Choose China  Chainese   Shanghai

ss8688, 2020-06-03:
Choose  China   Chinese   Shanghai

lurch, 2020-06-03:
Sounds like this might be referring to piwiz? ping @spl237 

@ss8688 Is this a problem specifically with the 64-bit Raspberry Pi OS, or does it also affect the 32-bit version?

spl237, 2020-06-03:
That sounds like piwiz; I can't think of any reason why language support should differ between 32 and 64 bit. It might just be that not very much of the GUI has had Chinese translations supplied, or that the Chinese fonts aren't installed.

ss8688, 2020-06-03:
Armhf vertion work well.Only affect arm64 version

spl237, 2020-06-03:
Just tried the settings above, and I got what looked like Chinese to me - 
[image]

There is an error when the wizard installs some of the Chinese fonts, but even with that, I was getting Chinese characters in the menus, as shown.

spl237, 2020-06-03:
Ah - sorry, reading again I think the OP was referring to the install error message, not that Chinese didn't actually work! 

I'll see if I can find out why those packages are not available in 64-bit; it looked to me that they were dependent on some Qt libraries which don't exist for 64-bit.

spl237, 2020-06-03:
OK, I know what the problem is; it's an issue with PackageKit trying to install both an armhf and an arm64 version of a package; I had to fix the same issue in Recommended Software. I'll port that fix across to the wizard.

spl237, 2020-06-03:
Fixed - will be included in next update of wizard.

Can someone with the relevant permissions please close this issue?

closed 2020-06-03 by ss8688

ss8688, 2020-06-03:
Close now