raspberrypi/Raspberry-Pi-OS-64bit #35: Java OpenJDK 8 LTS not available for download

brucelong, 2020-06-02:
I am running a old version of a minecraft server, and need Java 8. However, it does not appear to be present on the repos. 
`
brucelong@test-Server:/var/minecraft-modded1710/server $ sudo apt install openjdk-8-jre-headless
[sudo] password for brucelong:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package openjdk-8-jre-headless is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'openjdk-8-jre-headless' has no installation candidate
`

lurch, 2020-06-02:
Looks like that package is only available in `stretch` and `sid` https://packages.debian.org/search?keywords=openjdk-8-jre-headless whereas Raspberry Pi OS uses `buster` packages.

brucelong, 2020-06-02:
Yeah, sorry. Had a weird memory of running it on amd64, but I must be misremembering.

closed 2020-06-02 by brucelong