raspberrypi/Raspberry-Pi-OS-64bit #36: SSH Install - password is not raspbian

hitechrv, 2020-06-02:
All my fun stuff like microHDMI cables is locked up at work. I just got my P4 8gig.

I installed the 64bit beta using the new foundation installer. I added an SSH file to the boot directory and got SSH to prompt me for a password. Is their not one because Debian does not enable root by default? Or is is different than raspbian since this is no longer using raspbian?

pelwell, 2020-06-02:
The standard, out-of-the-box password is `raspberrypi`.

pelwell, 2020-06-02:
I mean `raspberry`.

lurch, 2020-06-02:
Indeed:
https://www.raspberrypi.org/documentation/remote-access/ssh/README.md
https://www.raspberrypi.org/documentation/remote-access/ssh/windows.md

hitechrv, 2020-06-02:
raspberry is what I should have entered. It is the first thing I tried. None of the above work. I will double check the links provided.

C:\Users\pc>ssh 192.168.50.159
The authenticity of host '192.168.50.159 (192.168.50.159)' can't be established.
ECDSA key fingerprint is SHA256:ERdXAv38O/z3dsLIbNXE6xaNBnLJhXOIxG1dlmdQkpY.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.50.159' (ECDSA) to the list of known hosts.
fna1\pc@192.168.50.159's password:
Permission denied, please try again.
fna1\pc@192.168.50.159's password:
Connection closed by 192.168.50.159 port 22

C:\Users\pc>ssh 192.168.50.159
ssh: connect to host 192.168.50.159 port 22: Connection timed out

C:\Users\pc>ssh 192.168.50.159
ssh: connect to host 192.168.50.159 port 22: Connection timed out

C:\Users\pc>ssh 192.168.50.159
ssh: connect to host 192.168.50.159 port 22: Connection timed out

C:\Users\pc>ssh 192.168.50.159
ssh: connect to host 192.168.50.159 port 22: Connection timed out

C:\Users\pc>ssh 192.168.50.159
ssh: connect to host 192.168.50.159 port 22: Connection timed out

C:\Users\pc>ssh 192.168.50.159
fna1\pc@192.168.50.159's password:
Permission denied, please try again.

pelwell, 2020-06-02:
You need to tell ssh that you want to log in as user `pi`: `ssh pi@192.168.50.159`

hitechrv, 2020-06-02:
Thanks!!

pi@192.168.50.159's password:
Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.
Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed May 27 18:13:56 2020

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@raspberrypi:~ $

closed 2020-06-03 by ghollingworth