raspberrypi/Raspberry-Pi-OS-64bit #37: ssh access installing headless

nettuno77, 2020-06-03:
using same procedure of ever, the username "pi" has a different password

ghollingworth, 2020-06-03:
No it doesn't, it's still "raspberry"

Do you get:

pi@192.168.1.144's password:
Permission denied, please try again.

nettuno77, 2020-06-03:
yes, I've tryed many times... but no luck. Also attempt different passwords... but still no luck

nettuno77, 2020-06-03:
ok, I try to type slowly, very slowly, and get login.
Then I logout and try many times to login typing password at different speed, but every time I logged in successfully. Buh, I don't know, so we can close this issue. Strange behaviour.

pelwell, 2020-06-03:
It just worked for me with a clean install. I'd close this if I could.

closed 2020-06-03 by ghollingworth

nettuno77, 2020-06-05:
No, nothing, it continue to give me Permission denied....
But I'm trying with a pi 3B+, I don't know if it matters