raspberrypi/Raspberry-Pi-OS-64bit #38: Where is the source code?

ossguy, 2020-06-03:
I'd like to build Raspberry Pi OS 64-bit just as I'm able to build NOOBS from source:

https://github.com/raspberrypi/noobs#how-to-rebuild-noobs

In particular, I want to know where I can find all the source code needed to build the image described at https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370 , which would include the instructions needed to do the building (similar to what's at the above link for NOOBS).

Could you point me to where I can find this source code, including instructions?  Thanks!

lurch, 2020-06-03:
https://github.com/RPi-Distro/pi-gen

ossguy, 2020-06-03:
Hmmm, I don't see any specific instructions for building the 64-bit version there.  Is that a setting?  In any case, this should probably be documented in https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/blob/master/README.md since this is the repository where we're told to report issues.

XECDesign, 2020-06-03:
https://github.com/RPi-Distro/pi-gen/issues/413

lurch, 2020-07-19:
[This comment](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/9#issuecomment-660668554) suggests that pi-gen can now build 64-bit versions?