raspberrypi/Raspberry-Pi-OS-64bit #39: Kodi Not Available to install

RafalNode, 2020-06-04:
Kodi  is not available. When I try to install it,

```
root@raspberrypi:/home/pi# sudo apt-get install kodi
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
Nie udało się zainstalować niektórych pakietów. Może to oznaczać,
że zażądano niemożliwej sytuacji lub użyto dystrybucji niestabilnej,
w której niektóre pakiety nie zostały jeszcze utworzone lub przeniesione
z katalogu Incoming ("Przychodzące").
Następujące informacje mogą pomóc rozwiązać sytuację:

Następujące pakiety mają niespełnione zależności:
 kodi : Wymaga: kodi-bin (>= 2:18.6-2~buster) ale 2:17.6+dfsg1-4+b1 ma zostać zainstalowany
        Wymaga: libshairplay0 ale nie da się go zainstalować
        Poleca: libva-intel-vaapi-driver ale nie da się go zainstalować
        Poleca: libva1 ale nie da się go zainstalować
E: Nie udało się naprawić problemów, zatrzymano uszkodzone pakiety.
```

Karnel:
```
root@raspberrypi:/home/pi# uname -r
5.4.42-v8+
```

Tested on a rpi4 8gb

camccar, 2020-06-04:
I second this.

andrum99, 2020-07-09:
I third this. The problem is that only the Debian version of Kodi is available, and it refuses to install due to dependency issues. There needs to be an arm64 version of the package on archive.raspberrypi.org, like there is for armhf.

eclipse95, 2020-08-11:
Hi,
As workaround, you can recompile the package by uncommenting sources repositories and with:
```
sudo apt-get build-dep kodi
sudo apt-get install libdrm-dev  # missing building dependency
apt-get source -b kodi
sudo dpkg -i kodi*.deb # this line can be improve
```

andrum99, 2020-08-11:
> Hi,
> As workaround, you can recompile the package by uncommenting sources repositories and with:
> 
> ```
> sudo apt-get build-dep kodi
> sudo apt-get install libdrm-dev  # missing building dependency
> apt-get source -b kodi
> sudo dpkg -i kodi*.deb # this line can be improve
> ```

That doesn't really help - there's no source for the kodi package on `archive.raspberrypi.org`, so the only option available is the Debian version, which runs too slow to be useful on the Pi, since it is completely unaccelerated.

elnono, 2020-08-30:
kodi seems to be available in arm64 but not libshairplay-dev which are only in armhf

maybe adding libshairplay-dev in arm64 in rapsberry os repos will unlock the install

eclipse95, 2020-09-01:
Hi,
@elnono, The package ``Kodi`` doesn't have ``libshairplay-dev`` as dependency. It doesn't need to be added. Moreover, ``libshairplay-dev`` and ``libshairplay0`` are already available in ARM64 Debian Package (Buster backports).

@andrum99, VLC is not accelerated (#52 ), so it's not better. You could try to rebuild Kodi ARMHF package from Raspberry Pi OS repo with ARM64 libraries and stuff.

tlaurion, 2020-09-02:
@eclipse95 

Same error: `kodi : Depends: libshairplay0 but it is not installable`

```
ubuntu@ubuntu:~$ grep -Rv "#" /etc/apt/sources.list* 
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal main restricted
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-updates main restricted
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal universe
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-updates universe
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal multiverse
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-updates multiverse
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-backports main restricted universe multiverse
/etc/apt/sources.list:deb http://archive.canonical.com/ubuntu focal partner
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-security main restricted
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-security universe
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-security multiverse
/etc/apt/sources.list.d/team-xbmc-ubuntu-ppa-focal.list:deb http://ppa.launchpad.net/team-xbmc/ppa/ubuntu focal main

```

```
ubuntu@ubuntu:~$ sudo apt install kodi
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kodi : Depends: libshairplay0 but it is not installable
        Recommends: i965-va-driver but it is not installable
        Recommends: kodi-peripheral-joystick but it is not installable
        Recommends: kodi-inputstream-adaptive but it is not installable
        Recommends: kodi-inputstream-rtmp but it is not installable
E: Unable to correct problems, you have held broken packages.

```

```
ubuntu@ubuntu:~$ sudo apt list --installed | grep kodi

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ubuntu@ubuntu:~$ 
```

tlaurion, 2020-09-02:
@camccar @eclipse95 @elnono @andrum99 
https://www.ubuntuupdates.org/package/xbmc/focal/main/base/libshairplay0
> This package belongs to a PPA: Xbmc

https://launchpad.net/~team-xbmc/+archive/ubuntu/ppa/+packages?field.name_filter=libshairplay0&field.status_filter=published&field.series_filter=
> No matching package for 'libshairplay0'. 

Who/where to tag?

lurch, 2020-09-02:
"Linux ubuntu ..."

This is the repo for the 64-bit Raspberry Pi OS. Any problems with Ubuntu-on-Pi will need to be reported to Ubuntu.

tlaurion, 2020-09-02:
Reported here http://disq.us/p/2blu6ag

jvonau, 2020-09-03:
> Reported here http://disq.us/p/2blu6ag

try here: https://bugs.launchpad.net/~team-xbmc

tlaurion, 2020-09-04:
> > Reported here http://disq.us/p/2blu6ag
> 
> try here: https://bugs.launchpad.net/~team-xbmc

Reported to team-xbmc@lists.launchpad.net, can't do bug report over there even with account.
Email:
```
Team-XBMC: please provide libshairplay0 for aarch64 (rpi4)

ubuntu@ubuntu:~$ sudo apt install kodi
Reading package lists... Done
Building dependency tree      
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kodi : Depends: libshairplay0 but it is not installable
        Recommends: i965-va-driver but it is not installable
        Recommends: kodi-peripheral-joystick but it is not installable
        Recommends: kodi-inputstream-adaptive but it is not installable
        Recommends: kodi-inputstream-rtmp but it is not installable
E: Unable to correct problems, you have held broken packages.

ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.4.0-1015-raspi #15-Ubuntu SMP Fri Jul 10 05:34:24 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux

ubuntu@ubuntu:~$ grep -Rv "#" /etc/apt/sources.list*
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal main restricted
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-updates main restricted
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal universe
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-updates universe
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal multiverse
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-updates multiverse
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-backports main restricted universe multiverse
/etc/apt/sources.list:deb http://archive.canonical.com/ubuntu focal partner
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-security main restricted
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-security universe
/etc/apt/sources.list:deb http://ports.ubuntu.com/ubuntu-ports focal-security multiverse
/etc/apt/sources.list.d/team-xbmc-ubuntu-ppa-focal.list:deb http://ppa.launchpad.net/team-xbmc/ppa/ubuntu focal main

ubuntu@ubuntu:~$ sudo apt list --installed | grep kodi

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ubuntu@ubuntu:~$

From https://www.ubuntuupdates.org/package/xbmc/focal/main/base/libshairplay0

    This package belongs to a PPA: Xbmc

https://launchpad.net/~team-xbmc/+archive/ubuntu/ppa/+packages?field.name_filter=libshairplay0&field.status_filter=published&field.series_filter=

    No matching package for 'libshairplay0'.

Thanks
```

XECDesign, 2020-09-04:
This won't be fixed until there's an appropriate kodi build with the right acceleration options and patches enabled. It's not just a matter of dependencies.

edurenye, 2020-10-05:
This problem happens on Debian 'buster' and anything like Raspberry OS that uses 'buster' but it works on 'sid', so maybe an option would be to have a build with Debian 'sid'?

abn0mad, 2020-10-15:
Kodi is installable if you add the deb-multimedia teams repositories to your apt sources.

sudo echo "deb http://www.deb-multimedia.org buster main non-free" >> /etc/apt/sources.list
sudo echo "deb http://www.deb-multimedia.org buster-backports main" >> /etc/apt/sources.list

sudo apt update

sudo apt-key adv --keyserver hkp://pool.sks-keyservers.net:80 --recv-keys INSERT_MISSING_KEY_HERE

sudo apt upgrade && sudo apt install kodi

jackyaz, 2020-11-13:
> Kodi is installable if you add the deb-multimedia teams repositories to your apt sources. They provide arm64 / aarch64 binaries for many packages that are perfectly usable on the 64-bit beta of Raspberry Pi OS.
> 
> sudo echo "deb http://www.deb-multimedia.org buster main non-free" >> /etc/apt/sources.list
> sudo echo "deb http://www.deb-multimedia.org buster-backports main" >> /etc/apt/sources.list
> 
> sudo apt update
> 
> sudo apt-key adv --keyserver hkp://pool.sks-keyservers.net:80 --recv-keys INSERT_MISSING_KEY_HERE
> 
> sudo apt upgrade && sudo apt install kodi
> 
> NOTE: Depending on whether you're running stock Mesa or have compiled your own you may get the following errors:
> 
> libva info: VA-API version 1.4.0
> libva info: va_getDriverName() returns 0
> libva info: Trying to open /usr/lib/aarch64-linux-gnu/dri/vc4_drv_video.so
> libva info: va_openDriver() returns -1
> Failed to open VDPAU backend libvdpau_vc4.so: cannot open shared object file: No such file or directory
> 
> Despite those (possible) errors however Kodi does function properly and plays video nicely.

Do you get hardware accelerated video or audio passthrough?

andrum99, 2020-12-25:
Anotehr workaround can be found here -> https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=296126

camccar, 2021-01-19:
I was able to compile Kodi with some help from the kodi forums. I have not been able to get 4k h.265 file to play yet but if you are brave you can build it and 1080p h.264 files play fine. I'll work on getting it to build in a dockerfile this week for other's to try. 

https://forum.kodi.tv/showthread.php?tid=351221&page=3

camccar, 2021-01-20:
Here is a dockerfile that builds kodi and includes instructions for copying it from the container to the local filesystem for running. If someone is looking for a way to run kodi on raspberry pi os 64 bit this will compile it for you. Takes around 2.5 hours to compile. Right now 4k h.265 content is not playing idk why. 

https://github.com/camccar/rpi4-kodi-build-dockerfile

camccar, 2021-01-21:
I guess hardware acceleration just isn't available on 64 bit yet. Due to some MMAL error

df911, 2021-11-03:
Hello,
I have a Raspberry 4 with arm64 and I try to install Kodi v19.
is there a version for this architecture ?

JamesH65, 2021-11-03:
The 64 bit RPiOS is still a beta, some things will not yet be working. I'd suggest using Open or LibreELEC distros if you want to use Kodi.

popcornmix, 2022-02-02:
kodi should be available in repo for 32-bit and 64-bit RPiOS. @RafalNode can you test and close if it works for you?

RafalNode, 2022-02-02:
> can you test and close if it works for you?

Unfortunately, I have no way to verify, my device is damaged :(
Let someone else test it and I'll shut it down.

bigOconstant, 2022-04-01:
I just want to say I have tested kodi in the latest rpi 64 os and 4k content doesn't play. content that plays fine on librelec. Would be nice to be able to run a regular OS and use kodi at the same time.

popcornmix, 2022-04-02:
Read [this post](https://forums.raspberrypi.com/viewtopic.php?p=1935073#p1935073).
You need config.txt settings to enable hevc decoder and increase cma.