raspberrypi/Raspberry-Pi-OS-64bit #40: 64 Bit OS package builds

mikebakke, 2020-06-04:
I note the earlier thread regarding Kodi not being available and I also have found one not present which is MongoDB.  This is a real shame for me because the 32-bit version I've installed on my "production" Pi4 is ancient. This means really that I need to keep my Ubuntu 20.04 instance running pretty much (so far) just for that package as the repo there is current (ish). I did try flipping the 64bit switch on a Pi3 and although it installed, it then wouldn't run.

I don't really understand how this works - are package (re)builds triggered by Pi OS developers or is each individual package controlled separately by specific maintainers? 

If the latter, how do I find who to ask, and if the central team is there any info on timescales?

TIA

Mike

lurch, 2020-06-04:
My understanding ts the the 64-bit packages in Raspberry Pi OS 64-bit come directly from Debian arm64.
So does this help? https://packages.debian.org/search?keywords=mongodb&searchon=names&suite=stable&section=all

mikebakke, 2020-06-04:
That brings a list which matches what I see with an apt search mongo on the Pi 64 bit system - all those are additional modules required for other packages - the core mongodb-server isn't there. On my prod Pi4 with Buster installed, it is there but only the 32bit version.

I'm going to head over to Mongo's site and download the Arm64 deb from there - I try to avoid that so updates are easier but life ain't that easy ;-)

Thanks

lurch, 2020-06-04:
Seems Debian only packages `mongodb-server` for jessie and stretch, whereas Raspberry Pi OS uses buster
https://packages.debian.org/search?keywords=mongodb-server

(Which IMHO makes this a Debian issue rather than a Raspberry Pi issue :wink: )

mikebakke, 2020-06-04:
Technically you are correct, but it's my issue on a Pi :-)

And this is a problem with the proliferation and fragmentation issues on distros... My (imperfect) understanding is that RP and UB are "basically" Debian underneath so the divergences confuse me but as it's been this way for 20+ years I don't expect it to change imminently. Had hoped Snaps might bring relief but that seems like another war brewing.

pelwell, 2020-06-04:
But this isn't a fragmentation issue - Debian appears not to be supporting mongodb on Buster - it's more like a non-fragmentation issue where you would like us to do something non-standard at a time when we are trying to avoid that.

mikebakke, 2020-06-04:
If Debian don't support on Buster but Raspian (32 bit) and Ubuntu 64 bit do, it seems like fragmentation to me but as I say my understanding is no doubt imperfect. I have two choices it seems - either download the .deb from Mongo themselves or use Ubuntu forwards. My preference is to stick with Pi OS but ultimately that will be driven by that applications that are available - maybe it's different for you, but I run all these Pis for applications and what they do, not the OS. The reason I say my preference is Pi OS is to have the firmware as close as possible to the hardware AND the fact that NFS booting on Pi is both available and so simple. For now I'll run side by side and at some future point choose one.

Cheers

pelwell, 2020-06-04:
From what I remember, Raspbian retains the old packages even though the don't appear in the new release.

MongoDB's installation page for Debian (https://docs.mongodb.com/manual/tutorial/install-mongodb-on-debian/) gives special installation instructions for Buster - grab the .deb package from their own site, but it appears to be only built for amd64. There is a similar process for Ubuntu (https://docs.mongodb.com/manual/tutorial/install-mongodb-on-ubuntu/), only there they support a bunch of architectures including arm64. You need to ask MongoDB why they aren't supporting arm64 on Debian Buster.

lurch, 2020-06-04:
`apt policy mongodb-server` on Raspberry Pi OS 32-bit reports:
```
mongodb-server:
  Installed: (none)
  Candidate: 1:2.4.14-4
  Version table:
     1:2.4.14-4 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
```
And 2.4.x is approximately the same version that https://packages.debian.org/search?keywords=mongodb-server says was in jessie, and much older than the 3.2.x that was in stretch. https://www.mongodb.com/download-center/community says the latest version is 4.2.x

So if you'd rather trade convenience for security that's entirely your choice... :wink:

closed 2020-06-04 by mikebakke