raspberrypi/Raspberry-Pi-OS-64bit #41: Need password to mount USB stick/shutdown for user 'pi'

MarkDH102, 2020-06-05:
Pi4 4GB Sandisk 64GB. Fresh install. update && full-upgrade. Only added NoMachine as running headless and wireless.

I have just the one user (pi) as per the installation.
I have the file /etc/sudoers.d/010_pi-nopasswd (I've not modified it)
It has permissions 0440.
It contains the single line:
pi ALL=(ALL) NOPASSWD: ALL

In the file manager, I can see the USB stick but I get a "Not authorised to perform operation" when trying to open it.
When shutting down or rebooting, I get asked to enter the password for user pi.

"groups" shows :

pi adm dialout cdrom sudo audio video plugdev games users input netdev gpio i2c spi

lurch, 2020-06-05:
Is this different behaviour between the 32-bit and 64-bit versions of Raspberry Pi OS?

MarkDH102, 2020-06-05:
All of my Pi's currently run 32 bit BUSTER. And none of them have this problem. All set up as single user 'pi' out of the box.

XECDesign, 2020-06-05:
Are you also using NoMachine on the 32bit setup as well? It sounds like a policykit issue, which I've sometimes seen with remote desktop software.

MarkDH102, 2020-06-05:
No,I use RealVNC, however it is not yet available in 64 bit form.
I'm not a linux expert so unfortunately policykit means nothing to me. But I will look it up and see if it can be worked around.

lurch, 2020-06-05:
I think @XECDesign is suggesting that perhaps installing NoMachine has changed the security policies or something? See if you can reproduce the problem using a display attached to the Pi _before_ installing NoMachine? :man_shrugging:

XECDesign, 2020-06-05:
I don't think nomachine changes anything, I think it runs a slightly different kind of environment. But I never had to look too closely as policykit, so I don't really know. I just know I've seen this with other similar software.

Maybe give this a go - https://www.nomachine.com/TR05N06836

MarkDH102, 2020-06-05:
Ok, I've looked at
/usr/share/polkit-1/actions/org.freedesktop.login1.policy
And changed
org.freedesktop.login1.power-off
<allow_any>auth-admin-keep</allow_any>
to
<allow_any>yes</allow_any>
And this allows me to shutdown without a password.
And I've done the same to the reboot entry.

NOTE: On my 32 bit Pi's the entry still says <>auth-admin-keep</>

Not sure which one to modify for the mount of a USB stick but I'll get there eventually...

Thanks very much for the pointer XECDesign.

MarkDH102, 2020-06-06:
Ok, there is some serious weirdness going on with NoMachine. My screen is a Debian screen not a Raspberry Pi one. I've now installed x11vnc and see the "proper" raspberry Pi OS screen and the shutdown and USB mount/eject all work. I'm not an expert so I have no idea what is going on. But my advice is not to install NoMachine…
I modified the UDisks2.policy file for the mounting of devices etc. but could not get the USB mount/eject to work without needing a password.

closed 2020-06-06 by MarkDH102