raspberrypi/Raspberry-Pi-OS-64bit #43: Jupterlab cannot be installed， show network has not arrived

wujianzhang1237, 2020-06-05:
Pip3 install jupyterlab.Is it because the default source does not have a corresponding resource?

lurch, 2020-06-05:
Same underlying issue as #42 ?

bennuttall, 2020-06-05:
Yes, please see the links I've provided in #46.

Borillion, 2020-06-08:
> Pip3 install jupyterlab.Is it because the default source does not have a corresponding resource?

Work around, build from source :)

vb0, 2020-07-03:
https://www.sagemath.org/ got built well too which is pretty impressive (as it's basically beside its own code a very large collection of various tools/programming languages). I could never build it before on Raspbian (and not because of pure memory limitations). Beside the current/preferred Jupyter Notebook interface it can [pip install jupyterlab](https://ask.sagemath.org/question/42876/other-than-jupyter-notebook-is-there-any-other-way-to-use-sage/?answer=42877#post-id-42877) and it seems to be working well (I'm not so familiar with it but I might actually start using it now, it looks nice).