raspberrypi/Raspberry-Pi-OS-64bit #44: PI4 ARGON ONE CASE INCOMPATIBILITY

valenroy, 2020-06-05:
There are errors with the GPIO interface and it prevents the pi 4 8GB model from booting. It boots during initial installation of the 64bit OS but it would run into random instances after a reboot that it would not boot again. This happens after an update & upgrade cycle randomly. I have reinstalled the OS countless to narrow down the issue but there's no clear indication what could be causing the issue.

The pi 4 works fine without the case.

lurch, 2020-06-05:
> The pi 4 works fine without the case.

Then this sounds like an issue for the case-maker to resolve, rather than Raspberry Pi? :shrug:

valenroy, 2020-06-05:
It is working fine with the 32-bit version so I thought this could be a relevant issue with the 64-bit OS when it comes to the GPIO interface. Sorry to mention that.

ghollingworth, 2020-06-05:
It's possible there's something different with the version 5.4 kernel.  Can you do an rpi-update on the 32bit version and see if it also happens there?

CodePosse, 2020-06-29:
32bit is fine, 64bit was not, until... I tried the fan but not the daughter/expansion board. I am not sure the way things "wake up" but after doing a little more plug/unplug it is fine just like my other fan hats. Also tested: IceTower and ADAFruit LED PiHat

mi-hol, 2020-06-29:
Initially I had intermittently no HDMI output using a cheap MiniHDMI cable with this case.
I turned out to be caused by mechanical tolerances of the involved MiniHDMI connectors, preventing a **reliable** connection. 

Solution:
I've cut down the plastic isolation on side of the cable to get the plug firmly seated.
Maybe that is the reason in your case too. Not getting output on connected monitor can easily be interpreted as a "not booting device" when in fact it is.

linareyne, 2020-07-02:
> 32bit is fine, 64bit was not, until... I tried the fan but not the daughter/expansion board. I am not sure the way things "wake up" but after doing a little more plug/unplug it is fine just like my other fan hats. Also tested: IceTower and ADAFruit LED PiHat

Are you saying that it still won't work for you with the expansion board that extends ports to the back of the case? Does the power button work fine still?

ceboxsell, 2020-07-02:
Works perfectly for me on my pi 4 4gb. script installed and ran, power button works, fan works no issue found. Am running with i2c devices from the Argon one and all ok.

junhaoliao, 2020-10-10:
Please try rpi-update. I was having the boot issue as well but everything is now fine after the update. 

> There are errors with the GPIO interface and it prevents the pi 4 8GB model from booting. It boots during initial installation of the 64bit OS but it would run into random instances after a reboot that it would not boot again. This happens after an update & upgrade cycle randomly. I have reinstalled the OS countless times to narrow down the issue but there's no clear indication what could be causing the issue due to the random nature after an update/upgrade.
> 
> The pi 4 works fine without the case.

WZ9V, 2021-01-09:
If I do rpi-update on an 8GB that was booting (without Argon case) it stops booting.  I can confirm that an 8GB Pi 4 and the Argon case works with latest 32-bit OS via imager but does not with latest 64-bit 8/20/2020.

ceboxsell, 2021-01-09:
I can confirm it does work on pi4 8 GB with latest updates. 
No problems found for me.

WZ9V, 2021-01-09:
> I can confirm it does work on pi4 8 GB 64 bit with latest updates.
> No problems found for me.

Do you have the latest case (V2) with the full size HDMI ports?

I tried both the 5/27/2020 and 8/20/2020 and neither would boot when plugged into the Argon One case.  Latest 32-bit works just fine for me.

ceboxsell, 2021-01-09:
No. It wasn't clear that is the one you are using so can't comment on it. 
Do you have the argon script installed ? There might be a library conflict.

WZ9V, 2021-01-09:
I don't get that far it won't even start after the imaging with the 64-bit, It looks like it starts then at some point the green LED goes off and the red one is left on.  I've been doing this headless (since it usually works) so that is all the feedback I have now.  I could try a serial console but other than imaging all I do is add ssh and a wpa_supplicant.conf both of which work fine with 32-bit and 64-bit without the argon case.  This is the newer V2 style and it works perfectly including booting the M.2 SATA drive in 32-bit.  So I don't think it is the case.

WZ9V, 2021-01-10:
So here are my latest findings.+

The original style Argon One with two micro HDMI connectors works no problem.

The new style Argon One with full size HDMI will not boot with the 64-bit images as downloaded.  It will boot the 64-bit image if you have done an apt update apt full-upgrade cycle to get the latest updates.  So if you install and update the SD card with a Pi 4 outside of the case, you can get it to work so I expect and updated 64-bit beta image might work directly once it is released.

ceboxsell, 2021-01-10:
Nice work. Thanks for the update good to know.

cheesyfries7, 2021-02-04:
Just wanted to add, I flashed the latest 64-bit image (2020-08-24) on an SD card and it was able to boot in my Argon M.2 case on my first try. Did all the updates for good measure and now it seems to be running fine, no problems even after rebooting.

dbck, 2021-03-12:
I have the Argon ONE M.2 Case for Raspberry Pi 4 8GB and running a full upgraded raspios (installed using raspios_arm64-2020-08-24.img) and I'm not able to get any hdmi port working. Is this a known issue? Should it work?