raspberrypi/Raspberry-Pi-OS-64bit #47: Package libgfortran3:armhf is not available, but is referred to by another package.

varungujjar, 2020-06-06:
Hey Pi Guyz

been trying to install some python scripts and i get this error.. been 3 days but with no luck..  could use a little help here.. :(

ImportError: libgfortran.so.3: cannot open shared object file: No such file or directory

have tried sudo-apt get install libgfortran3.
get this following error
Package libgfortran3:armhf is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libgfortran3:armhf' has no installation candidate

I tried to force install a stretch package libgfortran3 but still that didnt help..
kind of lost and could use a little help..

lurch, 2020-06-06:
If you're using the 64-bit version of Raspberry Pi OS, `apt-get` should be attempting to install the arm64 version of a package, not the armhf version?

But anyway, it looks like Debian don't package `libgfortran3` for Buster https://packages.debian.org/search?keywords=libgfortran3

Looks like these packages **do** exist in Buster though:
https://packages.debian.org/search?keywords=libgfortran4
https://packages.debian.org/search?keywords=libgfortran5
:shrug: 

This is probably something you'll need to chase up with the author of the "python scripts" you're trying to use?

varungujjar, 2020-06-06:
okay thankx a lot.. :)

closed 2020-06-07 by varungujjar