raspberrypi/Raspberry-Pi-OS-64bit #48: Garbled screen when running an OpenGLESv2 app on plain X11 without lightdm

imbens, 2020-06-06:
This has been discussed in [https://github.com/raspberrypi/firmware/issues/1382](https://github.com/raspberrypi/firmware/issues/1382) in a follow up to an unrelated camera issue. I now have more precise steps to reproduce the problem. It happens only when using kernel 5.4 (it does happen with the latest 32bit image when you rpi-update to the 5.4 kernel) and fkms and X11 without lightdm with a fullscreen Xwindow and with vsync enabled. The precise steps are as follows:

Start from the 64bit image from https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-05-28/2020-05-27-raspios-buster-arm64.zip
Boot the image, it will boot to the desktop.

In a terminal or via ssh enter (this fixes the raspberrypi-dev package):
```sudo apt update
sudo apt -q -y full-upgrade```
At this point you should probably reboot.
In a terminal or via ssh enter:
```sudo apt -q -y install libegl1-mesa-dev libgles2-mesa-dev libx11-dev xterm
git clone git@gitlab.com:imbens/opengltest
cd opengltest
gcc opengltest.c -lGLESv2 -lEGL -lX11 -o opengltest
export DISPLAY=:0 && ./opengltest```
The application will show a red rectangle on a background that cycles between green and blue.
In a terminal or via ssh enter (this will disable lightdm):
`sudo systemctl set-default multi-user.target`
Reboot. The pi will now boot to a console.
In the console or via ssh enter:
`sudo xinit &`
This will start plain X11 without a desktop manager and it will open an xterm.
In xterm or via ssh enter:
```cd opengltest
export DISPLAY=:0 && ./opengltest```
The application will show a garbled red rectangle.
Edit opengltest.c and change `eglSwapInterval (mEGLDisplay, 1);` to `eglSwapInterval (mEGLDisplay, 0);` (this will disable vsync).
In xterm or via ssh enter:
```gcc opengltest.c -lGLESv2 -lEGL -lX11 -o opengltest
export DISPLAY=:0 && ./opengltest```
Now the rectangle is ok again.

ghollingworth, 2020-06-11:
Is this specific to the 64bit version?

Otherwise this needs to be logged against the linux repo if it is generic

imbens, 2020-06-11:
No, it is not specific for the 64bit version, but the 32bit image is still using the 4.19 kernel. I will create an issue in the linux repo and I will close this one.

closed 2020-06-11 by imbens