raspberrypi/Raspberry-Pi-OS-64bit #49: display_rotate not working in 64bits

cecilcosta, 2020-06-07:
Forcing a VGA with hdmi_group=1 and hdmi_mode=1 (I have a very small screen) and setting display_rotate=1 makes no effect. The fisrt 1 attributes work fine, but display_rotate doesn't. I tried from 0 to 3.

lurch, 2020-06-07:
Does https://www.raspberrypi.org/documentation/configuration/display_rotation.md help?

cecilcosta, 2020-06-07:
xrandr works with GUI, but I'm more interested in rotating the terminal.
I also tried display_hdmi_rotate and display_led_rotate and they don't work.

lurch, 2020-06-08:
Did you miss the bit where it says "If you are using the console only (no graphical desktop) then you will need to set the appropriate kernel command line flags. Change the console settings as described on the this page." ?

I believe the 64-bit OS image might be only shipping with the KMS graphics driver, and doesn't include the Legacy graphics driver.

closed 2020-06-11 by ghollingworth

cecilcosta, 2020-06-12:
You are right, the main issue was that I had to remove the line _dtoverlay=vc4-fkms-v3d_

TheBlackfurGuy, 2021-11-20:
I have a similar issue. The option `lcd_rotate=2` only has an effect in the first few seconds of bootup. After that it switches around again. I assume at that point some breaking driver is enabled.

lurch, 2021-11-24:
@TheBlackfurGuy See this comment: https://github.com/raspberrypi/linux/issues/4686#issuecomment-967317257