raspberrypi/Raspberry-Pi-OS-64bit #50: Extra Software (& Chromium)

BusinessKid, 2020-06-08:
I've installed the 2020-05-27 64Bit beta on my RazPi 4 w/4G. Apart from /boot/config.txt specifying no kernel and  4 being supplied (I guessed kernel8) it seemed to work fine Booting WAS slow. I stuck in the sdcard, gave up on it booting, but had to leisurely trek out to the facilities (As I'm a stroke victim), and when I finally returned it was up. Wifi, Screen size, all came good. Sound came good once I managed to unmute the thing. I have 1xHDMI attached. That's pretty near gold standard for easy install.

I am in the habit of deleting cookies & history on browser exit. It's an option in browsers like Palemoon, but not in Chromium. Do I have to delete ~/.config/chromium? Are there other browsers on the list of 'to-do' things to add to this? 

I get it that there's very little software out there, and very few 64bit apps It would be lovely if

1. We could get at a list of stuff we could install with apt as debs become available.
2. Nearly all software can be installed " sudo make "DESTDIR=/somewhere install" and the direct6ory treeis lined up in /somewhere ready for packaging. Perhaps a script/python thing to
make debs? It's hardly done by hand. ldd is on basically every box, and someone who compiles his own could provide the patch, and the .deb for the beta only

lurch, 2020-06-08:
> Apart from /boot/config.txt specifying no kernel and 4 being supplied (I guessed kernel8) it seemed to work fine

There's no need to edit `config.txt`, it should all work "out of the box". Did it not work initially?

> it seemed to work fine Booting WAS slow

Maybe you have a particularly slow SD card? What does the Raspberry Pi Diagnostics tool say?

> I am in the habit of deleting cookies & history on browser exit. It's an option in browsers like Palemoon, but not in Chromium.

Perhaps using "incognito mode" is a workaround for that? :shrug: 

> We could get at a list of stuff we could install with apt as debs become available.

Everything that's been compiled for arm64 in Debian Buster is available...  go wild! :grinning: 
https://packages.debian.org/stable/

> Perhaps a script/python thing to make debs?

I believe that's a question for the upstream Debian project, rather than Raspberry Pi.

BusinessKid, 2020-06-09:
Booting 1st time was very slow. I may have timed out on it while it was arguing with itself. The second time I got distracted, so it had time to settle whatever argument there was.

No sense using incognito - some (paranoid) sites recognize their own cookies and weird stuff happens. Can't you delete cookies?

Everything that's been compiled for debian buster (stable) is squat. Didn't I make that clear? Sorry.

lurch, 2020-06-09:
> No sense using incognito - some (paranoid) sites recognize their own cookies and weird stuff happens. Can't you delete cookies?

I'm sure you probably can, but it's not something I've tried myself. Probably a question for the [forums](https://www.raspberrypi.org/forums/) rather than the 64-bit Raspberry Pi OS beta issues-list.

> Everything that's been compiled for debian buster (stable) is squat. Didn't I make that clear? Sorry.

Hmmm? Can you give an example of the sort of thing you're looking for that "we could install with apt as debs become available" that isn't already pre-packaged in upstream Debian Buster? :confused:

BusinessKid, 2020-06-09:
> Hmmm? Can you give an example of the sort of thing you're looking for that "we could install with apt as debs become available" that isn't already pre-packaged in upstream Debian Buster? confused

What I said holds. There's hardly any aarch64 stuff.There isn't even the OS. I can't get VNC - the notes say it's a WIP (= broken compile, waiting for maintainer to get off the beer). There's a list of stuff that isn't working announced with the download. I can't even get another browser - firefox or palemoon are not on your repos. In my use case (media box in front of TV with a high NEGATIVE score for it's interface) I would immediately install network printing, VNC,  with maybe a webcam & microphone as I can't hump things around anymore and the tv has the front room.

And all these Debian  project names paralleled with version numbers just adds confusion for Debian newbies.  What version number is Buster? What version number is Sid? You're not as bad as Ubuntu who clearly take moments of insanity by their project (typical 'prancing pickle' or 'flutterhouse fugitive'), No wonder every ubuntu user sticks to numbers!

The .img install is great, by the way. Pretty much the only way to go with the RazPi, where booting can be such an absolute pain in the sphincter.  BTW when I had dd'ed it accross fine, I noticed you had some little tool and I was supposed to transfer it across with that. But I imagine that just invokes dd :-/.

pelwell, 2020-06-09:
I don't understand the purpose of your post. You've installed our 64-bit preview, to a slow card by the sounds of it, then started listing all the things it can't do. If the limitations of this WIP are too restrictive for you then I suggest you install something else and come back when we announce the next update.

The tone of your messages including the insult to the maintainers makes it sound like you just needed to vent a little, because it won't achieve anything other than waste the time of the people who are hard at work trying to improve things.

BusinessKid, 2020-06-09:
I'm sorry if I gave that impression. That was not what I hoped to do. 

I don't particularly need to vent anything. In my search for an OS for this , I've had (and still have) Noobs-32bit, but wanted a 64 bit, largely for the full FPU support. I've tried Fedora's offering; but it is uninstallable on my kit. Slackware Arm 32 bit, which works but is tedious to set up and fussy to install; slarm64 (SLackware ARM64) which has good resources, it seems, but not enough directed development to get anywhere. It is also so security conscious the user is locked out! Then you offered the beta which is fine, but supporting software is slim.

I know what I want. Frankly I want the options as they come along, and I realize they're not there yet. But I don't want to undertake compiling behemoth programs every time I want to use one. I've done enough of that in my time. I'm going to close this because I largely have my answers. I have nailed down that Debian isn't offering other betas.  So I'll use yours, be happy with it, and wait.

closed 2020-06-09 by BusinessKid