raspberrypi/Raspberry-Pi-OS-64bit #51: seccomp causes tracker to crash

trabant-rgb, 2020-06-08:
The new 5.4.42-v8+ #1319 has seccomp enabled, which causes the tracker-extract process (an file indexing tool that comes with Gnome) to crash.

```
** Message: 21:28:34.366: Starting to process 239 items
** Message: 21:28:34.366: Extracting metadata for 'file:///home/pi/Music/The%20Connells%20-%20'74%20'75%20(acoustic%20cover).mp3'
** Message: 21:28:34.367: MIME type passed to us as 'audio/mpeg'
** (tracker-extract:7872): DEBUG: 21:28:34.368: Loading seccomp rules.
Bad system call
```
Is there any way to disable seccomp, i.e. per sysctl.conf?

ghollingworth, 2020-06-11:
First Google hit gives this, looks related,

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=851148

closed 2020-06-11 by ghollingworth

trabant-rgb, 2020-06-11:
Well, it's not exactly that. The version of tracker-extract I am running is the one from the actual Debian repositories, 2.1.6, not the one from the 2016 bug report you mentioned  (1.10.3-1). Also, tracker-extract does not lead to an unresponsive system, nor does it core dump. It just exits repeatedly with the mentioned error message.

Before going to this 64Bit kernel + userland, I had a current 32 Bit Raspbian with kernel 4.19.119 and the same version of Gnome installed. Tracker-extract was running fine then... thus I thought it might be related to the new 5.4 kernel's config and/or 64 bit system calls. 

Both issues seem to be seccomp related though. Do you by chance know if the kernel config has changed regarding to that option?