raspberrypi/Raspberry-Pi-OS-64bit #52: VLC does not make use of the builtin H264 decoder

trabant-rgb, 2020-06-08:
When playing videos via VLC, the Raspberry builtin H264 encoder is not used, so the performance and quality goes down.

pelwell, 2020-06-08:
If you've read the Forum post about the 64-bit OS (and you really should: https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370) this is literally the first Known Issue.

trabant-rgb, 2020-06-08:
> If you've read the Forum post about the 64-bit OS (and you really should: https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370) this is literally the first Known Issue.

You are right, and I actually did read it. Nevertheless I thought an "official" issue would be ok, so I'd be kept posted about this when there's a solution.

trabant-rgb, 2020-06-17:
One interesting thing though... the Gnome-delivered "totem" player seems to be able to address the H264 decoder, at least bcmstat shows so... however, totem is performance-wise so bad the video gets stuck almost immediately. But that totem quality thing is neither a new thing nor 64 bit related, it was as bad with 32 bit.

kralq, 2020-06-27:
On the desktop experience side the Unofficial 64-bit Ununtu https://jamesachambers.com/raspberry-pi-4-ubuntu-server-desktop-18-04-3-image-unofficial/ is miles ahead of the official Ubuntu images with:
- VLC/smplayer hardware accelerated video decoding
- 3D video hardware acceleration
- Working Bluetooth
- Raspberry Pi userland utilities included (vcgencmd, dtoverlay, etc.)
- raspi-config utility included

Seems that they are focused on the server side of the image and don't pay much attention to desktop features like bluetooth and hardware accelerated video.

What I didn't expect is that it would also be ahead of the Raspberry Pi Foundation 64-bit image even when it is still in beta.

I can confirm that with James A. Chambers' unofficial 64-bit image (64-bit 4.19.97 kernel) VLC has hardware accelerated video decoding. The installed VLC version is 3.0.8-0ubuntu18.04 arm64 build from the ubuntu repository.

The kernel configuration for the 64-bit kernel from the rpi-4.19.y branch as far as I can see from https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/blob/master/BuildPiKernel64bit.sh is in https://raw.githubusercontent.com/sakaki-/bcm2711-kernel-bis/master/conform_config.sh and https://raw.githubusercontent.com/TheRemote/Ubuntu-Server-raspi4-unofficial/master/conform_config_jamesachambers.sh

ghollingworth, 2022-02-02:
Out of date

closed 2022-02-02 by ghollingworth