raspberrypi/Raspberry-Pi-OS-64bit #57: No ethernet

Yajo, 2020-06-16:
My ethernet cable is plugged in, but it won't work.

If I plug it into the laptop, it works fine, so the cable is OK.

It was working in 32bit raspbian.

Yajo, 2020-06-27:
> Check for link lights?
> Checked network preferences?

Yes. No lights. About the preferences, what do you want me to show you? 

> Reverted Pi to 32-bit to retest?

No, but it was installed before and working fine.

fishd72, 2020-06-27:
Apologies if I’m covering stuff you already know, but in IT troubleshooting when you make a change, and something stops working unexpectedly, a good test is to revert the change to see if it recovers. This way, you isolate the fault to whatever you did in the change.

If you revert the Pi back to 32-bit, and your Ethernet _does_ start working again, then there is an issue with the _config_ in your 64-bit somehow. If you have a spare micro-SD Card for the 32-bit OS then this is likely the quickest test. 

I say here that it’s a config issue because:

1. As networking is a core feature, the Raspberry Pi foundation will have tested ethernet functionality before releasing the beta
2. I have Ethernet working just fine here under the 64-bit OS on my Raspberry Pi 4
3. You so far are the only person to post on here about ethernet issues 

So the balance of probabilities suggest that there is no general ethernet issue with the 64-bit version of the OS and it’s a local hardware or config issue.

If your ethernet port _doesn’t_ start working again under 32-bit then there is likely a hardware issue somewhere.

Following the OSI model for networking, it’s good to start at the physical layer.

1. Check the cable is securely connected at both ends of the connection; if the retaining clip is damaged, replace the cable with a new one to ensure it remains connected correctly during testing.
2. Try a known working cable connecting the same device to the same physical port on the network device you are connecting to
3. Try the suspect cable with a known working device and known working network port

The above should rule out the most common physical issues.

Moving onto the Linux config, [this](https://www.redhat.com/sysadmin/beginners-guide-network-troubleshooting-linux) seems to be a good article for troubleshooting the Linux networking config, so I’d suggest working through the steps there.

esbeeb, 2020-07-15:
My wired ethernet is working just fine in RPi OS 64bit, on my RPi 4.  Nothing special was needed for me.  Can't reproduce the error in the OP.

ryan763, 2020-07-21:
I installed normal image and 64bit image on RPi 4B 8gb and 2gb without issue here.

fishd72, 2020-07-21:
Unless @Yajo has further evidence / issues, propose this is closed?

Maynardnaze, 2020-07-21:
64 bit ethernet stopped working for me as well. went for vacation on 4th of july. came back on monday to no ethernet. the power did go out when i was gone.i have link lights. wifi works fine. i was using an nvme drive so i switched to a copy on an sd card. same thing. i get 169.x.x.x address. about to try again.

Maynardnaze, 2020-07-22:
also used an rpi 4 on the same cable and it works fine. honestly starting to think its a hardware issue. i have no idea what else it can be. big waste of a 8gb rpi

pelwell, 2020-07-22:
A DTS change just went into rpi-5.4.y that might have a bearing on this. There is likely to be an rpi-update release in the next 24 hours which will include that change.

Maynardnaze, 2020-07-22:
> A DTS change just went into rpi-5.4.y that might have a bearing on this. There is likely to be an rpi-update release in the next 24 hours which will include that change.

The problem lies with the 8gb rpi4. 4gb works fine on same cable. All brand new installs. Wifi does work 

32 bit fails to connect 
64 bit fails to connect

ryan763, 2020-07-22:
I am currently on newest code and have zero problems with a 2gb 4B now in 32bit and a 8gb 4B on 64bit working perfectly.
“

Both of my 4B rpi are using the official USB-C power supply with 3A.

Any chance this could also be a lack of power issue, in the event it is not a coding/software bug?

Maynardnaze, 2020-07-22:
> I am currently on newest code and have zero problems with a 2gb 4B now in 32bit and a 8gb 4B on 64bit working perfectly. “ Both of my 4B rpi are using the official USB-C power supply with 3A. Any chance this could also be a lack of power issue, in the event it is not a coding/software bug?

im using a canakit 5.1v 3.0 amp psu. honestly think its the actual board. ive used different rpi still nothing. ive tried multiple installs. different cables. 32 bit and 64 bit fail to connect.

klara31, 2020-12-07:
I have the same issue, running on a Pi4-2Gb. Every now and then at boot, ethernet does not work. Leds at the ethernet port are on, but no response on a ping. I am using the official power supply. No problems with the same board when running 32bit

lurch, 2020-12-07:
> Leds at the ethernet port are on, but no response on a ping.

That sounds like you're trying to use another (external) computer to ping the Pi4, to see if it's responding to ethernet ping requests? I suspect this problem will be **much** easier to diagnose if you're able to attach a screen and keyboard to the Pi, so that you can still "do stuff" on the Pi when the network is non-functional :wink:

klara31, 2020-12-07:
True, but I do not have a keyboard/mouse/screen attached to the pi (headless) as it is in a location where I do not have those available... 😉

klara31, 2021-01-22:
Update: I figured out what caused my ethernet issues, after changing cables and even the board itself. 

I am running docker on the pi. Docker has its own DHCP server serving the containers, but this seems to interfere with dhcpcd. I added ```denyinterfaces veth*``` to ```/etc/dhcpcd.conf``` and so far the problems are gone.

tablatronix, 2021-01-29:
@klara31 solution fixed mine, did not have issue on 32bit

Tachi107, 2021-12-17:
So can this issue be considered fixed?

closed 2021-12-17 by pelwell