raspberrypi/Raspberry-Pi-OS-64bit #61: Predictable network names in raspi-config

bowguy, 2020-06-19:
The interface in raspi-config gives the opposite results than as expected.   In "Enable Predictable Network Names?" selecting Yes gives the coded interface names.  Selecting No gives eth0 and wlan0.
If raspi-config is the same for all OS then you can close.

lurch, 2020-06-20:
From what I can tell, this **is** working as expected, it's just that your understanding is opposite :wink: 

https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/ says:
> this means that as soon as multiple network interfaces are available the assignment of the names "eth0", "eth1" and so on is generally not fixed anymore and it might very well happen that "eth0" on one boot ends up being "eth1" on the next.

So in fact eth0 and wlan0 _are_ the unpredictable names.

bowguy, 2020-06-20:
You're right, I misunderstood this.  Since I have always had 1 ethernet and one WiFi I interpreted this backwards.

closed 2020-06-20 by bowguy