raspberrypi/Raspberry-Pi-OS-64bit #62: error while loading shared libraries: libX11.so.6: cannot open shared object file: No such file or directory

Nikotine1, 2020-06-20:
When running wxtoimg, I get the following error:
```wxtoimg: error while loading shared libraries: libX11.so.6: cannot open shared object file: No such file or directory```

Yet, I have installed both libx11-dev and libx11-dev:i386:
```
pi@RPi4-sdr:~ $ sudo find / -name libX11.so*
/usr/lib/aarch64-linux-gnu/libX11.so
/usr/lib/aarch64-linux-gnu/libX11.so.6
/usr/lib/aarch64-linux-gnu/libX11.so.6.3.0
/usr/lib/i386-linux-gnu/libX11.so
/usr/lib/i386-linux-gnu/libX11.so.6
/usr/lib/i386-linux-gnu/libX11.so.6.3.0
```

Wxtoimg is running fine on a RPi3 32 bit.

lurch, 2020-06-21:
It seems that `wxtoimg` isn't [packaged by Debian](https://packages.debian.org/search?keywords=wxtoimg&searchon=names&suite=stable&section=all), so maybe you need to chase this up with wherever you downloaded wxtoimg from? :shrug: 

Oh, and the 32-bit arch for RaspberryPi is `armhf`, not `i386` ! :laughing:

Nikotine1, 2020-06-22:
Thank you @lurch , for pointing me towards ```armhf```.
I was able to install the missing libraries from that arch and now it's working!

The source code of wxtoimg is no longer available.
Someone has scraped together what he could find here: https://wxtoimgrestored.xyz/beta/
I don't think it can be recompiled for 64 bit.

For other Googlers, I had to do:
```
sudo dpkg --add-architecture armhf
sudo apt update
sudo apt install libx11-6:armhf
```
This resolved the error around libX11.so.6.
I got two more errors so had to install two more libraries:
```
sudo apt-get install libxft2:armhf
sudo apt-get install libasound2:armhf
```
After this wxtoimg started.

closed 2020-06-22 by Nikotine1

1165048017, 2021-02-20:
`
apt-get install gconf-service libasound2 libatk1.0-0 libc6 libcairo2 libcups2 libdbus-1-3 libexpat1 libfontconfig1 libgcc1 libgconf-2-4 libgdk-pixbuf2.0-0 libglib2.0-0 libgtk-3-0 libnspr4 libpango-1.0-0 libpangocairo-1.0-0 libstdc++6 libx11-6 libx11-xcb1 libxcb1 libxcomposite1 libxcursor1 libxdamage1 libxext6 libxfixes3 libxi6 libxrandr2 libxrender1 libxss1 libxtst6 ca-certificates fonts-liberation libappindicator1 libnss3 lsb-release xdg-utils wget
`