raspberrypi/Raspberry-Pi-OS-64bit #64: GPIO LCD driver for 64 bit OS

openoms, 2020-06-24:
Did anyone make the 3.5" GPIO display http://www.lcdwiki.com/3.5inch_RPi_Display work with the 64bit Raspberry OS yet?

Related issue: 
https://github.com/goodtft/LCD-show/issues/212

tux1c, 2020-08-14:
https://github.com/tux1c/wavesharelcd-64bit-rpi

kilrau, 2020-10-08:
> https://github.com/tux1c/wavesharelcd-64bit-rpi

Works like a charme if `/boot/config.txt` options are set correctly: https://github.com/tux1c/wavesharelcd-64bit-rpi/issues/3 !

closed 2020-10-08 by ghollingworth

kilrau, 2020-10-08:
Why close? The LCD still needs manual driver installation and fiddling with config options. It should work out of the box, until then this issue should stay open imho.

ghollingworth, 2020-10-08:
This display is a non-standard third party device, why do you think it should work 'out of the box'?  The instructions for installing it just need fixing on the originator's website

kilrau, 2020-10-08:
Alright.