raspberrypi/Raspberry-Pi-OS-64bit #65: v4l2loopback

kevindekemele, 2020-06-25:
I'm trying to use the v4l2loopback, which I have installed through the package manager.

However, when I try to use this module, I have to following error:
[image]

Any idea how to solve this problem?

pelwell, 2020-06-25:
How did you install v4l2loopback? Which commands did you run?

kevindekemele, 2020-06-25:
I did

´sudo apt-get install v4l2loopback-dkms´

and it said it was already installed.

pelwell, 2020-06-25:
And when you run `apt info v4l2loopback-dkms` it says?

kevindekemele, 2020-06-25:
This is the information

`pi@raspberrypi:~ $ apt info v4l2loopback-dkms
Package: v4l2loopback-dkms
Version: 0.12.1-1
Priority: optional
Section: kernel
Source: v4l2loopback
Maintainer: IOhannes m zmölnig (Debian/GNU) <umlaeute@debian.org>
Installed-Size: 115 kB
Depends: dkms (>= 2.1.0.0)
Suggests: v4l2loopback-utils
Homepage: https://github.com/umlaeute/v4l2loopback
Tag: admin::kernel, devel::lang:c, devel::library, hardware::video,
 implemented-in::c, role::devel-lib, role::program, role::source,
 works-with::video
Download-Size: 34.6 kB
APT-Manual-Installed: yes
APT-Sources: http://deb.debian.org/debian buster/main arm64 Packages
Description: Source for the v4l2loopback driver (DKMS)
 This module allows you to create "virtual video devices".
 Normal (v4l2) applications will read these devices as if they were ordinary
 video devices, but the video will not be read from e.g. a capture card but
 instead it is generated by another application.
 .
 This package provides the source code for the v4l2loopback kernel modules
 to be built with dkms.
`

kevindekemele, 2020-06-25:
Cloning the project on github: https://github.com/umlaeute/v4l2loopback
and `make` in the directory yields;

```
pi@raspberrypi:~/v4l2loopback $ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/pi/v4l2loopback modules
make[1]: *** /lib/modules/5.4.47-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```

pelwell, 2020-06-25:
`Source for the v4l2loopback driver` - the package is probably built on installation, but perhaps that failed. If you look in '/usr/src/v4l2loopback*' you should find the source code - probably the same as you found on GitHub, but it's worth trying to build it to confirm (you might need to use `sudo make`).

kevindekemele, 2020-06-25:
I have found the files in ´/var/lib/dkms/v4l2loopback´ and then source. However ´sudo make´ yields te same:

```
pi@raspberrypi:/var/lib/dkms/v4l2loopback/0.12.1/source $ sudo make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/usr/src/v4l2loopback-0.12.1 modules
make[1]: *** /lib/modules/5.4.47-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```

pelwell, 2020-06-25:
There's a good tutorial on building out-of-tree kernel modules for the Pi here: https://www.ploxiln.net/rpi_kernel_modules.html
Although it includes downloading the whole kernel source, you don't end up building that much of it.

pelwell, 2020-06-25:
But first, try install the kernel-headers package:
[ From  https://www.raspberrypi.org/documentation/linux/kernel/headers.md ]
```
$ sudo apt install raspberrypi-kernel-headers
```
If that doesn't work, try the other tutorial.

pelwell, 2020-06-25:
Ah - the lack of a functional headers package on the 64-bit OS is a known issue: https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4
However, if you scroll to the bottom of that thread you'll see that user @satmandu has published a script to create them, which you'll find here: https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51

kevindekemele, 2020-06-26:
Thanks. The scripts gave me the following error though:

```
+ mkdir -p /root/workdir/tmp/deb_YhkV/headers/boot/
+ cd /root/workdir/tmp/deb_YhkV/headers/usr/src/linux-headers-5.4.47-v8+
+ yes ''
+ make modules_prepare
make: gcc: Command not found
  HOSTCC  scripts/basic/fixdep
/bin/sh: 1: gcc: not found
make[2]: *** [scripts/Makefile.host:107: scripts/basic/fixdep] Error 127
make[1]: *** [Makefile:500: scripts_basic] Error 2
Makefile:659: include/config/auto.conf.cmd: No such file or directory
make: *** [Makefile:677: include/config/auto.conf.cmd] Error 2
```

If I do `sudo apt install gcc` it says it is already installed

These are the gcc folders/files /usr/bin

[image]

pelwell, 2020-06-26:
But does /usr/bin/gcc-4.7 exist? I think the highlighting in red means it doesn't. Try making it point to aarch64-linux-gnu-gcc-8 instead:
```
$ sudo ln -sf aarch64-linux-gnu-gcc-8 /usr/bin/gcc
```

ciffelia, 2022-06-12:
I managed to use v4l2loopback on Raspberry Pi 4 with 64bit OS.

```sh
curl -LO https://github.com/umlaeute/v4l2loopback/archive/refs/tags/v0.12.5.tar.gz
tar xzf v0.12.5.tar.gz && cd v4l2loopback-0.12.5
make && sudo make install
sudo depmod -a
sudo modprobe v4l2loopback devices=3

# Now I have 3 virtual video devices
v4l2-ctl --list-devices
```