raspberrypi/Raspberry-Pi-OS-64bit #69: Rainbow screen using POE on Pi4 w 4GB with bootable 240GB USB

mitchind, 2020-07-05:
No issue with normal powered Pi. But if I try and use the Pi POE power it comes up with rainbow screen
Pi 4 with 4GB RAM booting on Verbatim USB 240GB SSD - using powered USB 3.0 hub
Also attached is Coral USB Accelerator

Can I do some troublehooting here to help?

mitchind, 2020-07-05:
Booting normally using power card.

Also connected are USB keyboard and mouse on USB 2 ports
and HDMI monitor

$ vcgencmd version
Jun 1 2020 13:24:51
Copyright (c) 2012 Broadcom
version 6379679d1ec6a8c746d7e77e015f5b56b939976f (clean) (release) (start_x)

$uname -a
Linux picam 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux

$ sudo rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Mon 15 Jun 2020 01:36:19 PM UTC (1592228179)
LATEST: Mon 15 Jun 2020 01:36:19 PM UTC (1592228179)
FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000137ad
LATEST: 000137ad

ghollingworth, 2020-07-05:
We believe this is possibly an issue with GPIO1 on the PoE HAT...  Try rpi-update and this might fix the problem...

mitchind, 2020-07-06:
Okay - that's done
```
pi@picam:~ $ uname -a
Linux picam 5.4.50-v8+ #1324 SMP PREEMPT Wed Jul 1 17:13:08 BST 2020 aarch64 GNU/Linux
```

Now using the PoE hat, it's not recognizing the USB SSD boot drive on startup. Repeated error message loop that sdcard not found.
It still boots fine with USB-C power and SSD drive - just not with power from PoE hat

EEPROM version looks same so I assume I don't need to reapply that.

Is there some type of log I can pull up for more detailed help?

mitchind, 2020-07-06:
```
vcgencmd version
Jul  2 2020 15:00:30
Copyright (c) 2012 Broadcom
version 36c8be9515deddc9d2b1f469374f00d0a2df13f9 (clean) (release) (start_x)
```

5ft24dave, 2020-07-21:
The Pi POE hat is only rated 5V @ 2.5 amps.  Not enough for the Pi4 and an SSD. I'm running the UCTRONICS POE hat and my Pi4 4g and 8g both work fine and boot from my Samsung 860 EVO 500g SSD

RJDaPirate, 2020-09-04:
Did this issue ever get solved?  I cannot get a video input to work when using Raspberry OEM PoE Hat