raspberrypi/Raspberry-Pi-OS-64bit #70: HyperPixel4 dtoverlay does not load

Gadgetoid, 2020-07-07:
I've no doubt - given I'm not great with device-tree at the best of times - that there's something awry with my overlay and not necessarily Pi OS- but the HyperPixel 4 overlay does not load in the 64bit beta.

*Should* this dtoverlay still work? If not, does anyone have any hints for what the problem might be?

The dts source in question: https://github.com/pimoroni/hyperpixel4/blob/pi4/src/hyperpixel4.dts

With fragment 0 in place targeting the fb (I don't think this fragment is useful or necessary actually?) the dtoverlay will fail to load with:

```
[ 2377.500957] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/fb/pinctrl-names
[ 2377.500971] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/fb/pinctrl-0
```

If I remove this section it will fail with:

```
[ 2862.509217] OF: resolver: overlay phandle fixup failed: -22
```

I am able to forego the overlay and get the display working with the following settings:

```
gpio=0-25=a2
gpio=19=op,dh
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

Thank you!

pelwell, 2020-07-07:
Those "memory leak" warnings are just warnings, and they are unavoidable (who knows why - it doesn't seem like such a hard thing to track).

Why are you writing out your phandles explicitly? Did you start be decompiling an existing overlay? And how are you compiling it - if you aren't including `-@` in the dtc command line then you won't be generating the necessary fixups.

I recommend writing the overlay like this:
```
/dts-v1/;
/plugin/;
/ {
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&fb>;
        __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&dpi18_pins>;
        };
    };
    fragment@1 {
        target = <&gpio>;
        __overlay__ {
            dpi18_pins: dpi18_pins {
                brcm,pins = <0 1 2 3 4 5 6 7 8 9 12 13 14 15 16 17 20 21 22 23 24 25>;
                brcm,function = <0x6>;
                brcm,pull = <0x0>;
            };
        };
    };
    fragment@2 {
        target-path = "/";
        __overlay__ {
            rpi_backlight: rpi_backlight {
                compatible = "gpio-backlight";
                gpios = <&gpio 19 0>;
                default-on;
            };
        };
    };
    fragment@3 {
        target-path = "/";
        __overlay__ {
            i2c_gpio: i2c@0 {
                compatible = "i2c-gpio";
                gpios = <&gpio 10 0 /* sda */
                     &gpio 11 0 /* scl */
                    >;
                i2c-gpio,delay-us = <4>;        /* ~100 kHz */
                #address-cells = <1>;
                #size-cells = <0>;
            };
        };
    };
    fragment@4 {
        target = <&i2c_gpio>;
        __overlay__ {
            /* needed to avoid dtc warning */
            #address-cells = <1>;
            #size-cells = <0>;
            ft6236: ft6236@14 {
                compatible = "goodix,gt911";
                reg = <0x14>;
                interrupt-parent = <&gpio>;
                interrupts = <27 2>;
                touchscreen-size-x = <480>;
                touchscreen-size-y = <800>;
                touchscreen-x-mm = <51>;
                touchscreen-y-mm = <85>;
            };
        };
    };
    fragment@5 {
        target = <&ft6236>;
        __overlay__ {
            touchscreen-swapped-x-y;
        };
    };
    fragment@6 {
        target = <&ft6236>;
        __dormant__ {
            touchscreen-inverted-x;
        };
    };
    fragment@7 {
        target = <&ft6236>;
        __overlay__ {
            touchscreen-inverted-y;
        };
    };
    __overrides__ {
        touchscreen-inverted-x = <0>,"+6";
        touchscreen-inverted-y = <0>,"-7";
        touchscreen-swapped-x-y = <0>,"-5";
    };
};
```
and compiling it with:
```
$ dtc -@ -I dts -O dtb -o hyperpixel4.dtbo hyperpixel4
```
That compiles and loads without errors. The compatible string change is basically cosmetic, but I'm standardising on bcm2835.

Alternatively, you could hang the pinctrl nodes from the backlight node:
```
/dts-v1/;
/plugin/;
/ {
    compatible = "brcm,bcm2835";

    fragment@0 {
        target = <&gpio>;
        __overlay__ {
            dpi18_pins: dpi18_pins {
                brcm,pins = <0 1 2 3 4 5 6 7 8 9 12 13 14 15 16 17 20 21 22 23 24 25>;
                brcm,function = <0x6>;
                brcm,pull = <0x0>;
            };
        };
    };
    fragment@1 {
        target-path = "/";
        __overlay__ {
            rpi_backlight: rpi_backlight {
                compatible = "gpio-backlight";
                gpios = <&gpio 19 0>;
                default-on;
                pinctrl-names = "default";
                pinctrl-0 = <&dpi18_pins>;
            };
        };
    };
    fragment@2 {
        target-path = "/";
        __overlay__ {
            i2c_gpio: i2c@0 {
                compatible = "i2c-gpio";
                gpios = <&gpio 10 0 /* sda */
                     &gpio 11 0 /* scl */
                    >;
                i2c-gpio,delay-us = <4>;        /* ~100 kHz */
                #address-cells = <1>;
                #size-cells = <0>;
            };
        };
    };
    fragment@3 {
        target = <&i2c_gpio>;
        __overlay__ {
            /* needed to avoid dtc warning */
            #address-cells = <1>;
            #size-cells = <0>;
            ft6236: ft6236@14 {
                compatible = "goodix,gt911";
                reg = <0x14>;
                interrupt-parent = <&gpio>;
                interrupts = <27 2>;
                touchscreen-size-x = <480>;
                touchscreen-size-y = <800>;
                touchscreen-x-mm = <51>;
                touchscreen-y-mm = <85>;
            };
        };
    };
    fragment@4 {
        target = <&ft6236>;
        __dormant__ {
            touchscreen-inverted-x;
        };
    };
    fragment@5 {
        target = <&ft6236>;
        __overlay__ {
            touchscreen-inverted-y;
        };
    };
    fragment@6 {
        target = <&ft6236>;
        __overlay__ {
            touchscreen-swapped-x-y;
        };
    };
    __overrides__ {
        touchscreen-inverted-x = <0>,"+4";
        touchscreen-inverted-y = <0>,"-5";
        touchscreen-swapped-x-y = <0>,"-6";
    };
};
```

Gadgetoid, 2020-07-07:
Thank you- I think the answer to most of your questions can be explained by stating: "I still have virtually no idea what I'm doing." Device-tree is completely opaque to me and since I don't have the strength of will to trawl through documentation and tend to learn by example, it will probably remain so for some time. (It's not an easy thing to learn by example)

I suspect this overlay is adapted from the original HyperPixel overlay which was probably partially created by decompiling something else before I knew the location of the actual .dts files. That's a few years of baggage that's slowly crumbling into a mess.

I broadly understand the cosmetic change to "compatible" and the process of hanging pinctrl from the backlight node.

With any luck, actually having a servicable overlay that confirms to the naming conventions will finally give me something I can upstream- albeit there are still some edge cases preventing that.

I will try these tweaks!

pelwell, 2020-07-07:
I'll happily take questions on our Device Tree forum.