raspberrypi/Raspberry-Pi-OS-64bit #71: raspberrypi-kernel-headers mismatch with raspberrypi-kernel ... can't build out-of-tree kernel modules

pprindeville, 2020-07-09:
I booted the 64-bit image, then did:

```
% uname -r
5.4.42-v8+
% apt-get download raspberrypi-kernel-headers
...
% dpkg-deb -c raspberrypi-kernel-headers_1.20200601-1_armhf.deb | egrep '/lib/modules/.*/build'
lrwxrwxrwx root/root         0 2020-06-01 17:38 ./lib/modules/4.19.118+/build -> /usr/src/linux-headers-4.19.118+
lrwxrwxrwx root/root         0 2020-06-01 17:38 ./lib/modules/4.19.118-v7+/buil -> /usr/src/linux-headers-4.19.118-v7+
lrwxrwxrwx root/root         0 2020-06-01 17:38 ./lib/modules/4.19.118-v7l+/build -> /usr/src/linux-headers-4.19.118-v7l+
% 
```

so I downloaded the kernel again just to be sure...

```
% apt-get download raspberrypi-kernel
...
% dpkg-deb -c raspberrypi-kernel_1.20200601+arm64-1_arm64.deb | egrep '/lib/modules/[^/]*/$'
drwxr-xr-x root/root         0 2020-06-01 19:36 ./lib/modules/5.4.42+/
drwxr-xr-x root/root         0 2020-06-01 19:36 ./lib/modules/5.4.42-v7+/
drwxr-xr-x root/root         0 2020-06-01 19:36 ./lib/modules/5.4.42-v7l+/
drwxr-xr-x root/root         0 2020-06-01 19:36 ./lib/modules/5.4.42-v8+/
% 
```

Whaaaa....t?

Can someone please post the .deb for the kernel headers that actually matches the kernel?

lurch, 2020-07-09:
Duplicate of #4 ?

pprindeville, 2020-07-09:
Duplicate of #4

closed 2020-07-09 by pprindeville