raspberrypi/Raspberry-Pi-OS-64bit #72: Graphics desktop freezes

MSG-maniac, 2020-07-18:
Raspberry Pi 4 installs x64 system, use USB hard disk to boot
Download https://github.com/raspberrypi/firmware/releases/tag/1.20200601
It can be started after replacement, but it will cause the desktop to freeze and the mouse pointer to float.

mitchind, 2020-07-18:
Connected to what USB port? Powered by port or by powered USB hub?

MSG-maniac, 2020-07-18:
> 连接到什么 USB 端口？由端口供电还是由供电的 USB 集线器供电？

Hub power supply

MSG-maniac, 2020-07-18:
I'm not sure if it is caused by the 64-bit system or https://github.com/raspberrypi/firmware/. I have tried to replace many systems. As long as the USB SSD is used, this problem will occur because it needs to be replaced. The efi and dat files of the above project. Moreover, just causing the desktop to freeze will not cause the system to freeze, and the system load is normal.

MSG-maniac, 2020-07-18:
[71f4b57dc06c54a4a7b460605c4f8c87.zip](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/4942156/71f4b57dc06c54a4a7b460605c4f8c87.zip)

I recorded a video, you can check it out

MSG-maniac, 2020-07-18:
https://www.jeffgeerling.com/blog/2020/im-booting-my-raspberry-pi-4-usb-ssd

fishd72, 2020-07-21:
I am booting a Pi 4 8Gb model from a USB3 connected SATA M.2 SSD with no performance issues.

I'd suggest trying the 32-Bit OS and seeing how performance is there.
Alternatively, I do know that performance could depend on the USB > SATA convertor that you're using, I have a couple of different ones and I seem to recall one doesn't work very well whilst the other is fine, indicating a possible issue with some chipsets.

MSG-maniac, 2020-07-21:
I found the reason, because my mouse has set the report rate and it is not compatible with Linux. Restore the default USB report rate, just fine. Thank you both for your answers.

closed 2020-07-21 by MSG-maniac