raspberrypi/Raspberry-Pi-OS-64bit #73: install mariadb-server mariadb-client error

hiramvillarreal, 2020-07-19:
Hi, I tried to install mariadb-server mariadb-client with the next error:

The following packages have unmet dependencies:
 libgl1-mesa-dev : Depends: libgl-dev
 mariadb-client : Depends: mariadb-client-10.3 (>= 1:10.3.22-0+deb10u1) but it is not going to be installed
 mariadb-server : Depends: mariadb-server-10.3 (>= 1:10.3.22-0+deb10u1) but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

Regards.

esbeeb, 2020-07-19:
I installed the package "default-mysql-server", and that went off without a hitch.  But that's not MariaDB.

kaiomatico, 2020-07-21:
Have you done a `sudo apt-get update && sudo apt-get upgrade` first?

hiramvillarreal, 2020-07-22:
Thanks, I just made a new image and it works  thanks.

closed 2020-07-22 by hiramvillarreal