raspberrypi/Raspberry-Pi-OS-64bit #74: Bug

Heyman980, 2020-07-23:
I changed the gpu memory and when I restarted it, it refused to boot and gave me a bunch of errors.

pelwell, 2020-07-23:
Do you think it might be helpful to say which model of Pi you are running it on, with how much memory, and what you set gpu_mem to? Maybe?

Heyman980, 2020-07-23:
Pi4 4gb, I don't remember the gpu memory I set it to.

pelwell, 2020-07-23:
That's kind of important.

Heyman980, 2020-07-23:
I know...

Heyman980, 2020-07-23:
I erased the card and am trying to write the os again. I will try steadily increasing the gpu memory until the same crash happens. Then, I will copy the error message.

pelwell, 2020-07-23:
All memory that needs to be accessed by the VPU and some of the older hardware blocks has to fit into the first 1GB. On a Pi 4 we usually carve out 256MB for this shared area. I think things might start to struggle with gpu_mem much greater than 256 - 512 perhaps?

But why do you want a large gpu_mem on the 4? The graphics stack runs on the ARM and the improved graphics hardware can address any part of RAM for textures etc.

Heyman980, 2020-07-23:
I wanted it because it removed a small glitch in the output when scrolling.

Heyman980, 2020-07-23:
I moved it up to 720, it crashed again. The message is the attachment.

Heyman980, 2020-07-23:
[image]

pelwell, 2020-07-23:
1024 - 720 - 256 isn't a lot of space for anything else.

Heyman980, 2020-07-24:
Yeah

pelwell, 2020-07-24:
Like the overclock settings, gpu_mem is a power-user control that should be used with caution. Making it impossible to choose an invalid value without preventing you from using some valid values requires information that the firmware doesn't have or shouldn't need to care about. This is a "won't fix".

lurch, 2020-07-24:
https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md also clearly says "It is possible to set gpu_mem to larger values, however this should be avoided since it can cause problems, such as preventing Linux from booting."